RetroPie/RetroPie-Setup #791: Genesis Rom DIR not making Sega Genesis Menu come up

McPace, 2015-04-04:
I'm running the 3.0 beta image and I understood that creating a genesis dir in the roms would make the logo be the north american version of sega megadrive. Am I doing something wrong or is it broken? 

How do I get the Sega Genesis version on the menu?

Putting the roms in the megadrive dir brings up the sega menu and roms play fine. Putting them in the Genesis dir does nothing. I'd prefer the north american branding.

HerbFargus, 2015-04-04:
the genesis folder is only a symlink to the megadrive folder so it all ends up in the megadrive folder anyhow. you'll have to change the emulationstation theme reference in es_systems.cfg

/etc/emulationstation/es_systems.cfg

all i changed was the <theme> from megadrive to genesis- but I still kept all my roms in the megadrive folder. you can create a standalone genesis folder that isn't symlinked and then just change the rom path.. but its easiest just to change the <theme>

<system>
    <name>megadrive</name>
    <fullname>Sega Mega Drive / Genesis</fullname>
    <path>~/RetroPie/roms/megadrive</path>
    <extension>.smd .bin .gen .md .zip .SMD .BIN .GEN .MD .ZIP</extension>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ megadrive %ROM%</command>
    <platform>megadrive</platform>
    <theme>genesis</theme>
  </system>

McPace, 2015-04-04:
@HerbFargus Thanks, I'll give that a shot. 
I guess I was under the impression that 3.0 auto detected which rom dir they were in and displayed the theme based on that.

HerbFargus, 2015-04-05:
That is a really good idea as it would accommodate both sides of the world. Wouldn't be too difficult to do as the theme is already there, it would just need to be added as its own system in es_systems.cfg. The final say is up to Petrockblog though.

joolswills, 2015-04-05:
you shouldn't need to do anything. roms always live in the megadrive folder - but there is a genesis symlinks to that. the es config points to the megadrive folder. if you have removed the symlink and made a genesis folder this wont work of course.

joolswills, 2015-04-05:
if you want genesis branding, you need to switch the theme to that. I don't want to go back to having two rom folders really.

McPace, 2015-04-05:
Thanks! Got this squared away. 

Anyone else looking for this answer, it's also a good note that editing the file via your file browser gives a permission problem when you try to save the changes. So navigate there using an attached USB keyboard and use the command

sudo nano es_systems.cfg

to edit the file, then it doesn't bark at you about permission problems.

closed 2015-04-05 by McPace