RetroPie/RetroPie-Setup #793: per system retropie configuration being overwritten with settings

joolswills, 2015-04-08:
@gizmo98 @petrockblog 

Before we used to include the main config, and append additional ones. That means if a config was saved, it would overwrite the main config, so the appended config wouldn't be changed.

Now we are using the "include", settings are saved to the per system config, which is not what we want really. 

we can either: disable config saving on exit or  revert back to using the appendconfig again ? As we use the addsystem functionality, we can just put it in the function, so it wouldn't mean a lot of work to change.

thoughts ?

joolswills, 2015-04-08:
having all the retroarch config saved in every system retroarch.cfg is going to be unhelpful to users to maintain at least, so I prefer going back to using appendconfig I think, unless there is a better option?

petrockblog, 2015-04-12:
I also prefer to go back to using `appendconfig`. I do not see a better option for now.

joolswills, 2015-04-12:
I'll make the adjustment.

gizmo98, 2015-04-12:
If you enable save on exit for main config file all "system specific" options like paths will be stored there. Save on exit is really bad and can mess up the whole file.

joolswills, 2015-04-12:
hmm yeh, so it seems save on exit is a problem no matter whether we use the include or appendconfig.

@gizmo98 what do you suggest doing ?

It's a shame there isn't a nice way of having a global config, with the ability to save system specific stuff, without saying everything to one file (whilst still using the retroarch gui for configuration).

joolswills, 2015-04-12:
I believe what you are saying (please correct me if I'm wrong) is that with that option on, it's safer with our current method of using the include rather than append config, as at least configs will save to each system, rather than overwriting the global config ?

I shall hold off on this until I get some more detail from you as you may be more familiar with this than me.

gizmo98, 2015-04-12:
Yeah. But there is as well a problem if you enable save on exit in a system specific config file with main config included. After the first run "all" settings of the main and system specific file will be saved into the system specific file. From now on the included main config file will be ignored. 
The old ESConfig menu had a "start retroarch GUI without libretro core to setup main config" script.

joolswills, 2015-04-12:
we can add that back - but are you in favour of keeping the #include or using appendconfig? 

it seems neither is ideal!

gizmo98, 2015-04-12:
There were two reasons to remove appendconfig:
-Shorten command string. The command strings of the old modules was really long. Command string is short and readable now --> fixed!
-Don't mess up the main config file with system specific settings if save on exit is enabled.

Feel free to add appendconfig again.

joolswills, 2015-04-12:
your arguments are good though, so I'm happy to leave it as is- but we need to re-add the ability to adjust the main config as you mentioned so people can do main configuration there, and document that saving config when launching a specific emulator will save to the system config and override anything in the global config.

@petrockblog based on additional info from @gizmo98 , what do you think ?

petrockblog, 2015-04-12:
We should aim for the best user experience and go with the approach that could be described in one or two sentences to the users.
Having a way to explicitly edit the global configuration and another way to edit system specific files sounds like the way to go. I would not care for a long command string if we would in turn get a simpler way for editing global and system specific configurations.

taalas, 2015-04-25:
One other idea I had connected to appendconfig:

Currently we had 2 files affecting each system, the global configuration in all/retroarch.cfg and the system-specific one in <system>/retroarch.cfg. If a user edits either of those, the changes will be overwritten next time the installation is updated (by doing a binary install for example).

What if another 2 configuration files were added as user files and appended to the configuration, so that we would have the following scenario:

all/retroarch.cfg contains global settings that are set by RetroPie
all/retroarch.cfg.user contains settings a user can add/overwrite in global
<system>/retroarch.cfg contains local settings that are set by RetroPie
<system>/retroarch.cfg.user contains settings a user can add/overwrite locally

This would imho add the benefit of having a clearly defined space where users can add their configuration changes (e.g. shader settings) and be safe when updating their installation, since RetroPie-Setup will only overwrite the non-user files.

Risks:
- might be more complicated for users to understand at first
- invites problems if users have problematic settings in their configuration files and those won't get reset on installation/update, this might not be a problem though since one can always propose to delete those to test if the problems are rooted in those files

joolswills, 2015-04-25:
just to mention - the files are not overwritten on doing a binary install.

taalas, 2015-04-25:
Yes, I should have been clearer on that: What I meant was that the files could be defaulted to a state that is consistent with the current RetroPie standard installation. The goal for me would be to better separate user and distribution files.

What is the best way currently to revert the retroarch configuration to standards instead of reflashing a complete image?

joolswills, 2015-04-25:
well, you could delete the retroarch configs, and re-run the retroarch install

if you have an existing configs/all/retroarch.cfg any new base config config will be saved to configs/all/retroarch.cfg.rp-dist

taalas, 2015-04-25:
ah, that is handy to know, thank you

joolswills, 2015-06-20:
Resolved.

joolswills, 2015-06-22:
marked as resolved but it isn't really. 

So one of the issues with the new include method is: people think they can override retroarch.cfg options with the per emulator config options, but it seems retroarch takes the first config option it finds rather than the last, so we need to move the #include to the bottom it seems. 

@gizmo98 is this how you expect the behaviour or ? if we do move the include last, then when saving configs, will it put the configs after the #include line or before ?

Seems neither using  appendconfig nor #include is ideal - (not our fault, but poor logic in retroarch imho)

biscuits99, 2015-07-03:
Could it be kept as is, with a commented line in the system specific retroarch.cfg saying that overrides need to be above the #include? 

That said, controller overrides seem fine after the include - I'm guessing its just the retroarch settings themselves that need to be first.
Heres the reference:
http://libretro.com/forums/showthread.php?t=2733

A lot of users do get confused why RGUI settings dont get re-loaded for them though. Personally I really like the way it does it now, as it makes it difficult to really screw up your configs.

joolswills, 2015-07-03:
Have you confirmed putting them above does work as expected ? If so I'm happy to change the comment.

biscuits99, 2015-07-03:
I can confirm that putting
savestate_directory = /home/pi/RetroPie/
above the include line will override the entry I make in /all/retroarch.cfg

(I'm using beta 4)

If the line is added after the incude, the /all/retroarch.cfg takes precedence, as it was read/set "first".

I havent tested all retroarch.cfg settings, but it holds true for the ones I have checked.
Controller overrides seem fine after the include.

joolswills, 2015-07-03:
thanks - can you give me an example of a controller override ?

biscuits99, 2015-07-03:
Whilst the default exit emulator hokey is "Start" which is "7" on my controller.
If I add this line (either before or after the include)
input_exit_emulator_btn = "0"

It will happily use the "A" button as the exit emulator hotkey, in both instances disabling the "Start" key from having that function.

joolswills, 2015-07-03:
We don't have that in retroarch.cfg by default - so I'm not sure it is overriding as such ?

biscuits99, 2015-07-03:
I see it as the ES command runs the system based retroarch.cfg, which in turn pulls in the all/retroarch.cfg which in turn pulls in auto controllers. And I can override the controller settings by adding a line that would appear in them i.e. my controller file is "USB,2-axis_8-button_gamepad__.cfg"

This is really helpful with fba-libretro for example, as the "Select" button conflicts with some games, so I can add an override in the system based retroarch file for just that system.

Maybe my terminology is wrong?

joolswills, 2015-07-03:
well I was thinking along the lines of that anything in retroarch.cfg overrides the controllers perhaps ?

biscuits99, 2015-07-03:
Yes, that seems to be the case. As you say, perhaps not an override per se as its not set within retroarch, but a separate file. Just handy that it can be overridden as it gives some good flexibility.

joolswills, 2015-07-10:
from the commit:

```
retroarch uses the first setting it finds so when using an #include to include the main config, settings that should override those in retroarch.cfg should be above it.
    
adjust the per system retroarch.cfg generation, so the #include is placed at the end.
    
as before if a retroarch.cfg exists, no settings will be changed in it, and instead a retroarch.cfg.rp-dist will be used - however, if the existing retroarch.cfg is missing the all important #include line, it will be added, along with a comment at the top.
#793
```

new configs will now end up as (megadrive example)

```
# Settings made here will only override settings in the global retroarch.cfg if placed above the #include line

input_remapping_directory = /opt/retropie/configs/megadrive/
video_smooth = false
video_shader = /opt/retropie/emulators/retroarch/shader/phosphor.glslp
video_shader_enable = false

#include "/opt/retropie/configs/all/retroarch.cfg"
```

if someone has an existing config with the #include at the top, it won't be modified, but the default config will be put in retroarch.rp-dist

We can't really ensure that the config is upgradable from previous betas without possibly messing someones file, but updating from 2.6 works (it just adds the comment to the top, and the include to the bottom).

However upgrading fro 2.6 to 3.0 is not ideal as you get left with a lot of cruft behind.

joolswills, 2015-07-10:
here's the commit - https://github.com/RetroPie/RetroPie-Setup/commit/459151778650e570d0af6df6b45d96edf14369a6 please let me know if the changes I have described above seem ok to you.

biscuits99, 2015-07-10:
That looks good to me. Thanks!

joolswills, 2015-07-13:
thanks. closing.

closed 2015-07-13 by joolswills