RetroPie/RetroPie-Setup #794: GemRB Infinity Engine - Enhancement

amigagamer, 2015-04-08:
I'm in early stages of testing GemRB engine on retropie 3.0b/raspberry pi 2... If you'r not familiar with it, its a game engine to play Baldurs Gate, Icewind Dale, Planescape.

I'm working on a retropie module for it.  Tested with baldurs gate 1 (the GOG.com datafiles - copy all content to roms/gemrb/baldurs1 ).   GemRB itself hasn't had a lot of testing on RPI .... as for the build script- its my first attempt at one, so consider this highly experimental.

If theres any interest with the project Devs here please feel free to use any of the content for retropie.  https://github.com/amigagamer/gemrb-retropie

free5ty1e, 2015-04-08:
Awesome!!   I will also experiment with this :D

amigagamer, 2015-04-20:
Various fixes and improvements since original post.

Now creates configs/directories for Baldurs Gate 1+2 , Icewind Dale 1+2 , Planescape torment.
Briefly tested with all 5 games (character creation, intro movies, game start-movement,inventory,menus etc)
Configs in the config share for each game, including a sample config with all available options/explinations.

Forum thread http://blog.petrockblock.com/forums/topic/baldurs-gate-infinity-engine-available-for-testing

Game testers and script improvement suggestions most welcome!

free5ty1e, 2015-04-20:
Thanks!!  I'll have to check this out again this weekend and see if I can get some Baldur's Gate action going xD

ProxyCell, 2015-04-21:
i have two questions for you:
does this support a controller? (sounds unlikely given the games)
if it does, how can i exit the game with a controller?

amigagamer, 2015-04-21:
ProxyCell
No support for controllers sorry , really need a mouse and keyboard for this style of game

ProxyCell, 2015-04-23:
no problemo, i totally get it. i always ask though!

closed 2016-02-29 by gizmo98