RetroPie/RetroPie-Setup #795: Unable to 'Configure Retroarch Controller' / retropie 3.0 beta 2 / RPI 2B

rtrzebinski, 2015-04-09:
Hi there,

I've found issue related to version 3.0 beta 2 for Retropie 2.

Pi Model: 2B
RetroPie Version Used: 3.0 Beta 2
Built From: SD Image
Controller used: Rertolink SNES controller (worked perfectly with retropie 2.6 / raspberry pi 1 B+)
How to replicate the problem:

Keys X and Y are not being recognized in any Retroarch emulator (tested with playstation and n64).

I run 'configure retroarch keyboard/joystick' from 'retropie' emulation station menu a few times. Also I rebooted pi after that. No effect - only 'default' keys 'A' and 'B' can be used in games. 'X' and 'Y' can not be used. Running via command line (sudo ./retropie_setup.sh) takes no effect as well. As I see file /opt/retropie/configs/all/retroarch.cfg  is not being altered by this command (I use ls -l to check modification date). I believe that script is not saving it's result to this file.

Additional info:

Previously I was using Retropie 2.6 and 3.0 beta 1 on Raspberry pi B+ and it was working fine (the same gamepad) without any extra configuration (I didn't even have to configure retroarch controller). I guess this can be related only to Retropie 3.0 beta 2 for Raspberry pi 2. I also believe that (on my previous system) I was asked to press 'X' and 'Y' buttons by Emulationstation interface on first run, now it only wants me to press 'A' and 'B' - this can be possibly related.

If you need more details please let me know, I'm happy to help as much as I can.

Robert

ps. I've also created forum post here: http://blog.petrockblock.com/forums/topic/retropie-3-beta-2-issue-unable-to-configure-retroarch-controller/

swrrga, 2015-05-28:
I can confirm this bug, it is happening to me too.

The {D-pad, Start, Select, A, B} are all being recognized on my SNES-style controller. { X, Y, L, R } are not being registered in-game. I have run retroarch-config and the button inputs are recognized but they still won't work in-game.

dtminasian, 2015-05-28:
Same to me! I can't open map with x on earthbound. Took me ages to figure out why. I shouldn't've updated to a beta...

swrrga, 2015-05-28:
Following this advice seems to have fixed it for me:

"To keep it simple you could remove the existing .cfg files, just to make sure there is no conflict with (I think there *may* be an issue with the gamepad cfg file there by default but I’m just checking that out – by removing it as below it will get rid of the potential issue):

sudo rm /opt/retropie/emulators/retroarch/configs/*.cfg

Then create a new one with the setup script:
sudo /home/pi/RetroPie-Setup/retropie_setup.sh

And choose 3-Setup, 320-Configure Input for RetroArch

This shouldn’t really be changed though:
/opt/retropie/configs/all/retroarch.cfg 
"

Taken from http://blog.petrockblock.com/forums/topic/usb-snes-controller-d-pad-not-working-in-games/#post-97091

It seems there is indeed an issue with the default generic USB gamepad settings. The above commands simply remove all gamepad configs, then running the control script again to create a working file.

Super Metroid works! I can run in Super Mario!

dtminasian, 2015-05-30:
Thanks so much! a simple fix makes a huge difference! I'd hug you but this is the internet n stuff...

peterv26, 2015-06-07:
I also have the same problem, when i try to configure controls in Emulationstation I can do all the buttons except X and Y?  I have tried the advice above but no luck! 
It see's them in retroarch-setup and in the joystick test.
Any advice would be great, thanks.

biscuits99, 2015-06-10:
This may be a quicker fix - remove the problem file

sudo rm /opt/retropie/emulators/retroarch/configs/Retrolink_USB_SEGA_Saturn_Classic_Controller.cfg

joolswills, 2015-06-20:
This should be sorted now as no auto config files are installed by default - config for retroarch is generated after setting up a controller in emulationstation/

closed 2015-06-20 by joolswills