RetroPie/RetroPie-Setup #797: Logitech F310 Gamepad

jaybird1905, 2015-04-13:
Hey guys,

I've been digging around for weeks, watching videos, reading articles, fudging with config files ... and I'm still at a loss here.

On my Pi 2, I can't get my Logitech F310 Gamepad working for the life of me.

I've installed the xbox drivers, made them boot up at startup, etc. Still nothing.

When I go to the 'Configure input devices for RetroArch' option in RetroPie setup, it immediately exits back to the blue screen and splashes a 'success' type message.

I'm on the latest version of RetroPie and Raspbian.

Any help would be GREATLY appreciated!

Thanks,
Jay

jaybird1905, 2015-04-13:
So after getting a fresh copy of the latest RetroPie release, running rpi-update, and connecting my controller, going to ./retropie_setup.sh, and selecting the option for setting up a RetroArch controller, it immediately returns this message:

The configuration file has been saved as .cfg and will be used by RetroArch from now on whenever that controller is connected. 

This is without any xbox drivers installed. The strange thing is that my controller worked briefly when I first got my B+. It then stopped working completely. Now it doesn't work on my brand new Pi2 either.

I'm really at a loss here. It's recognized by my Macbook when I plug it in, so I'm pretty sure the controller is still functional.

aingold, 2015-04-13:
You're probably better off asking for support at the forums: http://blog.petrockblock.com/forums/forum/retropie-project-forum/

Tele42, 2015-05-01:
a F310 button mapping was just added to retroarch-joypad-autoconfig with this commit: https://github.com/libretro/retroarch-joypad-autoconfig/commit/181ca3f9df2cdc05dca9aeb5593365d20e1fd860. Build it from source to get this change immediately.

joolswills, 2015-06-20:
Please try with the latest beta which should configure the joypad automagically for retroarch.

joolswills, 2015-07-13:
Closing due to no feedback.

closed 2015-07-13 by joolswills

Giancarlos, 2017-10-14:
I had issues getting it working at all on a laptop prior to using the controller, be sure to check the bottom and set it from "X" to "D" or it wont work et all in some devices / platforms. Posting this tip in case someone falls upon this and finds this. I had a separate issue though.