RetroPie/RetroPie-Setup #799: 3do wrong path in Runcommand

HerbFargus, 2015-04-15:
Here is the error log:
```
RetroArch [ERROR] :: parse_config_file :: Couldn't find config at path: "/opt/retropie/configs/4do/retroarch.cfg"
RetroArch [WARN] :: rarch_log_libretro :: [libretro WARN] :: [4DO]: no system directory defined, unable to look for panafz10.bin
/opt/retropie/supplementary/runcommand/runcommand.sh: line 671:  4046 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-4do/4do_libretro.so --config /opt/retropie/configs/4do/retroarch.cfg /home/pi/RetroPie/roms/3do/Need\ for\ Speed,\ The\ \(1994\)\(Electronic\ Arts\)\(US\)\[A1115\ CC\ 735507-2\ R70\].iso --appendconfig /tmp/retroarch.cfg
```

`/opt/retropie/configs/4do/retroarch.cfg` needs to be `/opt/retropie/configs/3do/retroarch.cfg`

closed 2015-04-15 by HerbFargus

dockery, 2016-01-10:
ok, so how can I fix this?

joolswills, 2016-01-10:
this is not a current bug - it is closed. Make sure you are using a new retropie ? If you are having trouble I will need more information (you might want to try the forum first if you need support)

HerbFargus, 2016-01-10:
If you're trying to run it on a pi, you're wasting your time. It will never work well enough on a pi to be playable.