RetroPie/RetroPie-Setup #802: Rewind broken in latest 3.0 image, pi1 & pi2

free5ty1e, 2015-04-17:
Has anyone else had new problems with Rewind in all libretrocore emulators lately?  

I tried out the new 3.0 beta 2 image for both the pi1 and the pi2, and rewind is completely broken on both of them.  

I confirmed that I have enabled the rewind setting in each emulator's retroarch.cfg file and ensured at least a buffer size of 10 (have also tried 20 and 100), and I see the "Rewinding" on-screen display when I attempt to rewind, but it only jumps back one frame and continues as if I was NOT holding down the rewind button.

I was using rewind on plenty of emulators on the older 2.5 and 2.6 RetroPie images: NES, Sega Master System, Genesis, TG16, even SNES!  

What happened?  Any ideas?

ProxyCell, 2015-04-18:
you may wish to check the your retroarch.cfg as it changed up a bit since the older 2.x one. check and make sure it only has 1 line for enabling rewind

i'm currently compiling a new retropie and i will be able to test this rewinding issue for you - i honestly think i do a source build more than anyone else lol (up to 3 a day lol)

free5ty1e, 2015-04-18:
Nope, only one entry for rewind_enable in the configs/all/retroarch.cfg.  The on screen display for Rewinding comes up when the button is pressed, but it does not rewind.  I am using the combination SELECT + left on the R3 stick (PS3 controller), so this is mapped to an axis.  

I have confirmed that if I write the old RetroPie 2.6 image and copy the entire /opt/retropie/emulators/retroarch folder over to my latest RetroPie 3.0 SD card, rewind works great again.

This fixes rewind for both Pi1 and Pi2.  Retroarch's binaries from 2.6.0 actually work.  Retroarch from 3.0 is broken.

ProxyCell, 2015-04-21:
I have been unable to reproduce this. I compiled directly from source and rewind has continued to work for me.

free5ty1e, 2015-04-21:
I haven't tried building RetroArch from sources yet; I will try that as another fix.

joolswills, 2015-06-20:
Closing due to no feedback and feature is reported as working.

closed 2015-06-20 by joolswills