RetroPie/RetroPie-Setup #804: USB controller by retrolink not working while in game.

GreenLanternW, 2015-04-18:
I have run controller set up and every time an emulator is run "A" button will restart emulator. 
I have installed most reset script and run both controller setups. I'm a a loss for what is going on. controller works fine will in menu.

taalas, 2015-04-20:
It seems as if you somehow have a stuck menu toggle button somewhere. RetroArch provides the opportunity to define a button that when pressed in conjunction with another button will execute a RetroArch command. The standard autoconfig files will usually map this to the "Select" button. "Select" + "Start" will then exit the current Libretro core, "Select" + "A" will reset the current emulator. This might be what you are experiencing. How did you configure your input device?

GreenLanternW, 2015-04-21:
I configured the controller using the terminal and selected the option Configure Retroarch Controller. I also set up the controller within retropie. I;m running it on a rasberrypi 2.

cwittyman, 2015-05-27:
Same problem, not sure what im missing

joolswills, 2015-07-13:
I have a retrolink controller and I am unable to reproduce. I'll close for now, but please try with the latest beta and feedback if it's still an issue.

closed 2015-07-13 by joolswills