RetroPie/RetroPie-Setup #805: D-pad of x-box 360 controller not working ingame

jagiella, 2015-04-19:
After pulling the recent version of RetroPie-Setup and re-installing all cores, the d-pad of my x-box 360 controller (XEOX Pro Analog) is not working anymore ingame. In emulationstation itself everything works fine. I did several times a controller button reconfiguration in emulationstation as well as by the config script (./retropie_setup.sh), but it doesn't help. The result remains the same: ingame all buttons work except the movement directions (d-pad). I also tried to use the controllers analoge sticks, but without success...

What could be the issue? It seems like my configurations have no imact at all to the configuration the retroarch cores are using at the end.

HerbFargus, 2015-04-19:
I can second this issue- I'm having the same issue with my cheap USB gamepad.  These others seem to be having the same problem as well. This only happened in the last 3 months or so, before it worked just fine so something in the retroarch code must have broken (perhaps related to autoenable?) 

http://libretro.com/forums/showthread.php?t=1961

(BTW in the meantime until a fix is implemented upstream, this post on the Retropie forums is a workaround that should solve your problem.
http://blog.petrockblock.com/forums/topic/usb-snes-controller-d-pad-not-working-in-games/)

jagiella, 2015-04-19:
Indeed editing the file

`/opt/retropie/configs/all/retroarch.cfg`

and changing the following lines

`# Axis for RetroArch D-Pad. `
`# Needs to be either '+' or '-' in the first character signaling either positive or negative direction of the axis, then the axis number. `
`# Do note that every other input option has the corresponding _btn and _axis binds as well; they are omitted here for clarity.`
`# input_player1_left_axis = `
`# input_player1_right_axis = `
`# input_player1_up_axis = `
`# input_player1_down_axis = `

to

`# Axis for RetroArch D-Pad. `
`# Needs to be either '+' or '-' in the first character signaling either positive or negative direction of the axis, then the axis number. `
`# Do note that every other input option has the corresponding _btn and _axis binds as well; they are omitted here for clarity.`
` input_player1_left_axis = -0`
` input_player1_right_axis = +0`
` input_player1_up_axis = -1`
` input_player1_down_axis = +1`

Fixed the issue!

inverse, 2015-04-25:
I'll have to try this out as I am experiencing the same thing with a cheap usb controller

inverse, 2015-04-25:
Worked for me :+1:

inverse, 2015-05-25:
How to get this to work with two controllers though?

joolswills, 2015-07-13:
I believe this isn't an issue now? Please comment if the problem remains with the latest beta.

closed 2015-07-13 by joolswills