RetroPie/RetroPie-Setup #807: ES-Scraper not saving metadata edits

HerbFargus, 2015-04-22:
I know that this pertains to emulationstation but this issue may be retropie specific to emulationstation:

Quite a few people have mentioned their edits to metadata have not been saved:

http://blog.petrockblock.com/forums/topic/game-metadata-edits-not-saved/

http://blog.petrockblock.com/forums/topic/scraping-data-not-saved-in-es/

The only way I've been able to get manual edits to save is if I select quit emulationstation in the quit menu rather than shutdown or restart. Otherwise nothing is saved. Is there a way for edits to be written without needed a clean exit?

to reproduce- manually edit the metadata of a game within emulationstation and either shutdown or reboot- upon reboot it will be as if you made no changes.

gizmo98, 2015-04-22:
I think emulationstation (over)writes gamelists.xmls on exit. If you have not the permission to edit a  gamelist.xml emulationstation throws an error on exit.

HerbFargus, 2015-04-22:
Is there a specific way to check/get  permissions to edit gamelist.xml?

Tele42, 2015-04-22:
I reported this over at https://github.com/Aloshi/EmulationStation/issues/432.

gizmo98, 2015-04-23:
@herbfargus. This was only an example. Emulationstation seems to load all gamelists at startup and writes them back at exit. So if emulationstation is running and you modify a gamelist any modification will be overwritten.

Tele42, 2015-04-23:
I think @HerbFargus is saying manual entries via emulation station's gui, not directly editing the gamelist.xml files.

HerbFargus, 2015-04-23:
@Tele42 is right- I'm referring to manual edits via the emulation station GUI.

HerbFargus, 2015-05-10:
Closing as it is an issue best resolved through emulationstation.

closed 2015-05-10 by HerbFargus

x-manx, 2018-03-07:
Just restart the system instead of shutting it down.