RetroPie/RetroPie-Setup #815: DOSBox in RetroPie 3.0 Beta 2

dudleydes, 2015-04-30:
The config file, dosbox-SVN.cfg is not being written correctly in RetroPie 3.0. None of the changes (lines 74 to 77) in the DOSBox set-up script at https://github.com/petrockblog/RetroPie-Setup/blob/master/scriptmodules/emulators/dosbox.sh are being implemented. The most important is at line 74 where usescancodes is set to true by default so keyboard input is not correct.

Also, in line with my tutorials at http://dosonthepi.blogspot.co.uk/, would it be possible to change the emulators.cfg file to the following:

    dosbox="%ROM%"
    default="dosbox"
    rpix86="%ROM%"

That way, the shell scripts I suggest in the tutorials will work out of the box. Thanks.

joolswills, 2015-04-30:
There was an issue with dosbox that was fixed after the last 3.0 release. The problem was the folder permissions were initially created wrong so the config  wasn't being generated (so the options were not being set) see https://github.com/petrockblog/RetroPie-Setup/commit/fe4608f41fee050e40bbb9689413d5bc0a234716

joolswills, 2015-04-30:
please can you clarify on the second point you made - the config can't be set like that as we need to keep the two options for launching the emulators without the rom. the default is set so that launching the .sh scripts launches into commandline, but launching a rom will still launch it - but emulator can be switched. If you have an issue with this, please can you detail it.

joolswills, 2015-04-30:
regarding the first issue, if you delete "/opt/retropie/configs/pc", update the retropie-setup script and re-install dosbox it should correctly set the configuration up. the next 3.0 release will have the config dir set-up correctly also.

dudleydes, 2015-05-02:
The great feature of DOSBox is the mapper tool that lets me configure game controllers. Because of the different keyboard combinations possible, sometimes I need to create a unique mapper file for a game. This is where the issue arises.

Let's take Wolfenstein 3D as an example. The location of the unique mapper file for this game, wolf3d.map say, has to be given in the config file which, to avoid conflict with the default config file, I save as wolf3d.conf. For ease, I use a shell script (wolf3d.sh), to launch Wolfenstein 3D from ES. In the script I have added a -conf switch to the DOSBox command specifying the location of the config file, wolf3d.conf.

As it stands, running wolf3d.sh just takes me to the DOS prompt in DOSBox and the game does not run. The script will launch Wolfenstein 3D with the amendment suggested in my initial post but this does seem to stop mounting the roms folder in DOSBox when running the Start DOSBox script. The ideal solution would be where the Start DOSBox script and any bespoke script to launch games with their own config and mapper files are both accommodated.

I have updated the retropie-setup script and re-installed dosbox. The configuration is correctly set up. Thanks.

joolswills, 2015-05-06:
I can probably make it so it's possible to launch from a script will look into it.

free5ty1e, 2015-05-07:
That would be super awesome.  Perhaps if we choose an EXE or BAT or COM file to launch via EmulationStation, the DOSBOX launch script could look for a .CONF file with the same filename as the launched file, and if it exists utilize it?

Or I suppose even adding the capability to directly launch .CONF files appropriately from EmulationStation for DOSBOX would also be a possibility.  

This would allow me, for example, to map PS3 controls for Privateer, and then I could launch and play it from my couch at will without ever having to touch a keyboard :D

free5ty1e, 2015-05-07:
(I second this wholeheartedly -- I regularly utilize the awesome DOSBOX Mapper to create .conf files for playing games on a PS3 controller from my Mac.  Utilizing these on the RetroPie would be an amazing step forward for DOS emulation)

dudleydes, 2015-05-26:
G.A.M. Kertopermono has posted a script (below) at my <a href="http://dosonthepi.blogspot.com/2015/04/retropie-30-update.html?showComment=1432385806604#c9136972979763777716">DOS gaming blog</a> that will launch games in ES from an executable or from script.

    #!/bin/bash
    params="$1"
    if [[ "$params" =~ \.sh$ ]]; then
    bash "$params"
    else
    params+=" -exit"
    /opt/retropie/supplementary/runcommand/runcommand.sh 0 "/opt/retropie/emulators/dosbox/bin/dosbox $params" "dosbox"
    fi

It also launches the DOSBox command prompt but does not mount the roms folder. I don't know how much an issue this is. I can always type it in. In any case, I don't tend to use the DOSBox CL on my pi, preferring to use the desktop version to create config and mapper files or to install games before copying the files over to the pi.

I hope this is our solution.

joolswills, 2015-05-26:
With a few changes, the current shell launch script could be adapted to do a similar thing (but keep the mounting stuff). It's just a matter of my free time currently.

joolswills, 2015-05-31:
Have made some changes that should allow launching of .sh scripts based on the above suggestions (with some changes).

dudleydes, 2015-06-07:
Thanks a lot. The script works fine. I guess we can consider this issue closed.

closed 2015-06-07 by dudleydes