RetroPie/RetroPie-Setup #822: Audio in Mame4all

dudleydes, 2015-05-05:
I have been testing mame4all in Retropie 3.0 for Raspberry Pi 1 and found that some games (1942, '99: The Last War) had no audio. I checked the mame.cfg file and noticed that the sample rate was set to 44100. In my previous builds or Retropie, the sample rate was 22050 so I made this change and audio was restored.

I found out that the sample rate was set to 44100 as a result of issue https://github.com/petrockblog/RetroPie-Setup/issues/741. The audio in Donkey Kong is good at a sample rate of 44100 but admittedly poor at 22050. I tried the midpoint, a sample rate of 33075, and the audio in Donkey Kong is acceptable. This sample rate works for the games I mentioned above as well as games requiring samples (Space Invaders, Zaxxon).

Could the default sample rate for mame4all be set to 33075?

Tele42, 2015-05-05:
Excuse my ignorance, by why is 33075 the magic sample rate, not 32000? does the chipset even support 33075?

joolswills, 2015-05-05:
Perhaps you can detail your set up? mame4all has sound on 1942 with the samplerate at 44100 on my system. are you using hdmi audio ? 33075 is not any standard that I know of - maybe try 24000 or 48000 ?

joolswills, 2015-05-05:
you could also try

```
tvservice -d edid.dat
edidparser edid.dat > edid.txt
```

to get the edid info from your screen which should show the supported sample rates.

dudleydes, 2015-05-05:
Thank you for your responses.

After further testing, I have narrowed the issue to my HDMI monitor. The only sample rate that produces audio for 1942 is 33075, not 44100 nor 32000 even though these are supported and 33075 is not (see edid output for audio below)

    HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16 bits

I tried audio through the 3.5mm audio jack and I get audio at 44100. I'm going to close the issue as it's hardware-related.

closed 2015-05-05 by dudleydes