RetroPie/RetroPie-Setup #823: Lynx in RetroPie 3.0 Beta 2

craigerskine, 2015-05-05:
I've tested this many times and it has happened in other emulators, but I can get it to happen in the Lynx section at will now.

Starting from a fresh install of RetroPie 3.0 B2, if I launch, say, Awesome Golf it launches and plays fine for as long as I can stand. I can quit and launch the game many times without issue after the first few minutes of playing that particular game. However, after a few hours every time I try to launch Awesome Golf the Pi locks up on the terminal screen after exiting EmulationStation. I have to SSH in and reboot or disconnect power to reset.

The game will continue to never launch and lock up my Pi ad infinitum.

If I delete the two files that get created in the Roms folder whenever you launch a game (.rtc, .srm - see screenshot), the game launches and runs fine again for the next few hours. I'm not sure if it's a date/time issue since it takes a while for the game launching to stop working?

Again, I've had this happen in another game section as well, GameGear I think.

[image]

craigerskine, 2015-05-25:
I remedied this myself by changing the permissions on the atarilynx folder to prevent writing... now the .rtc and .srm files cannot be created so the game load normally more than just one time.

[image]

ProxyCell, 2015-05-26:
@craigerskine should this be done by a pull request?

craigerskine, 2015-05-26:
The ideal solution would be to prevent the Lynx emulator from creating the files in the first place. My solution prevents you from adding/removing more Lynx roms, which is not good. I'm not sure if there is a flag that can be set on the emulator to have this happen, or force the files to be created in a different folder (if that is even a valid fix).

closed 2016-03-26 by joolswills