RetroPie/RetroPie-Setup #824: GuiMenu.cpp?

geneworld, 2015-05-06:
Has GUIMenu.cpp changed to a new file? I am looking to edit the menu functions but I cannot find it. I installed it from source and used: 

git clone --depth=0 git://github.com/petrockblog/RetroPie-Setup.git

instead of:

git clone --depth=1 git://github.com/petrockblog/RetroPie-Setup.git

joolswills, 2015-05-10:
There is no GUIMenu.cpp as part of retropie itself, but perhaps some module. The second clone line is correct, but is completely unrelated. Perhaps you can detail the problem with a log as I'm not sure what your issue is at present.

geneworld, 2015-05-11:
Hi,
I am looking to edit the menu when you hit the start button and it gives you the options to exit, reboot, or shut down the system. In RetroPie 1.x it was guimenu.cpp but in 2.x ad now 3.x i am not sure where i can find these lines to edit.

joolswills, 2015-05-11:
you would need to look at the emulationstation source https://github.com/Aloshi/EmulationStation and modify that (and build a custom emulationstation too).

closed 2015-05-13 by joolswills