RetroPie/RetroPie-Setup #825: wicd prevents cpu speed scaling

Havelock-Vetinari, 2015-05-07:
wicd is on top and prevents cpu governor to scale frequency down. My Pi2 keeps running at overclocked 1GHz having over 55°C. When wicd is stopped it scales down to 200MHz and cools down to 40°C. Please make wicd installation/running optional.

Tele42, 2015-05-07:
not a problem with the default min freq of 600MHz

HerbFargus, 2015-05-10:
'sudo apt-get remove wicd-curses'. Problem solved.

diogomsantos, 2015-05-11:
Same opinion as Havelock-Vetinari. Wicd should be optional. The daemon uses almost 30MB of RAM and some extra CPU cycles which is a bad idea especially for Raspberry Pi 1. I use a bash a script to connect to wireless networks, its pretty simple but it does the job and doesnt need a daemon in the background consuming extra resources.

joolswills, 2015-05-12:
I will look into having it optional, and having maybe a simple gui config for wpa-supplicant for those that done want the daemon

HerbFargus, 2015-08-02:
with https://github.com/RetroPie/RetroPie-Setup/commit/148f8c0fd359b2f7efcc5897a8db57b59b410be0 this can now be closed

joolswills, 2015-08-02:
beat me to it ;-) efficiency!

closed 2015-08-10 by joolswills