RetroPie/RetroPie-Setup #826: /usr/bin/emulationstation: line 20: 2327 Segmentation fault $es_bin “$@”

Denus01, 2015-05-08:
I often get this error while exiting a game:
/usr/bin/emulationstation: line 20: 2327 Segmentation fault $es_bin “$@”

The system then keeps running in a loop, and I'm forced to cut off the power in order to restart the Pi.
I did several fresh installs, tried two different Raspberry Pi 2 's but I get the very same issues on both systems.

My two Pi 2 are over clocked on 1000/500/500 with overvoltage 2, but the temperatures are quite low, so I doubt that could be the issue.

[image]

joolswills, 2015-05-08:
Does it happen when not overclocked - did you test that before posting ? It could easily be the overclocking - you can't tell just from the temperature if you have a stable overclock.

Denus01, 2015-05-08:
It seems indeed to be a problem with the sdram on 500 which is obviously unstable on both systems.
Without overclocking, I don't get errors, and I'm now testing lower sdram speeds like 483: so far so good. 
Since the overclock setting was a standard one through the setup, and that I'm using heatsinks & aluminium cases, I couldn't even think about a possible issue with this.
Especially when I get only the very same error about "line 20 in ES" every time, and no other issues.

But thanks for pointing it out.
This request can be closed.

closed 2015-05-08 by Denus01

joolswills, 2015-05-08:
the error was just that emulationstation crashed - since it's launched from a shells script, you get the error on the line that launched it.