RetroPie/RetroPie-Setup #829: Xboxdrv and Retroarch autoconfig rework

taalas, 2015-05-11:
Hi,

I have been using the xpad driver in favor of xboxdrv for a while now since the RetroArch autoconfig for xboxdrv seems a bit incomplete in comparison to the kernel driver.

I have now finished my setup using xboxdrv and would like to suggest some changes (or would be happy if someone could clarify the choices made for this driver):

* use --dpad-as-buttons instead of --trigger-as-buttons
This is my main concern with the current state of the xboxdrv install script to be honest. Currently the dpad is mapped as an analog input device and correctly mapped in the appropriate autoconfig as input_player1_up_axis and such. Emulators like the SNES9x-next don't seem to like dpad inputs as axis though, at least in my case I can't get any input to be recognized. Same goes for the shoulder buttons (which in fact are analog inputs). Changing the driver call as proposed will change the dpad to digital buttons and the analog triggers to analog inputs.

* use an alternate autoconfig file
The autoconfig file for the Xbox Gamepad userspace driver would need to be replaced by a modified one (I have made one that has mappings similar to the kernel driver). This could either be achieved by overwriting the file after pulling the autoconfig directory or talking to the retroarch project about it.

The goal would be to achieve plug and play functionality without changing any input configuration files for Xbox Pads attached via xboxdrv.

joolswills, 2015-05-31:
@gizmo98 you have more experience than me with the joystick stuff - what do you think about the above ?

amigagamer, 2015-06-01:
Not retroarch related, but relavent to this topic , i removed --trigger-as-buttons due to being a big gamestream/limelight user.. ProjectCARS etc really need the analogue sensitivity on L2 + R2.  

For retroarch on retropie i cant think of any emulators i use that need L2 R2 so the change hasnt affected me, but would need testing perhaps with ps1 emulator or maybe some FBA/mame games use the triggers?
Not sure of the benefit of using dpad-as-button - it works either way.  Xbox pad as default seems to treat the dpad as analogue.

FWIW im using 2 360 wireless controllers with the usb wireless receiver.

sudo xboxdrv --daemon --id 0 --led 2 --dpad-as-button --silent --next-controller --id 1 --led
 3 --silent --dpad-as-button --dbus disabled --detach-kernel-driver &

my custom retroarch udev config as follows

input_device = "Xbox Gamepad (userspace driver)"
input_driver = "udev"
input_b_btn = "4"
input_y_btn = "6"
input_select_btn = "10"
input_start_btn = "11"
input_up_btn = "0"
input_down_btn = "1"
input_left_btn = "2"
input_right_btn = "3"
input_a_btn = "5"
input_x_btn = "7"
input_l_btn = "8"
input_r_btn = "9"
input_l2_axis = "+5"
input_r2_axis = "+4"
input_l3_btn = "13"
input_r3_btn = "14"
input_l_x_plus_axis = "+0"
input_l_x_minus_axis = "-0"
input_l_y_plus_axis = "+1"
input_l_y_minus_axis = "-1"
input_r_x_plus_axis = "+2"
input_r_x_minus_axis = "-2"
input_r_y_plus_axis = "+3"
input_r_y_minus_axis = "-3"

taalas, 2015-06-01:
@amigagamer Yes, that is exactly what I am doing on my setup currently as I detailed above. I also tried to start a conversation over at RetroArch but haven't had much success with getting feedback there.

taalas, 2015-06-01:
I also made a summary which buttons/axes map to which inputs comparing xpad, xboxdrv (currently), xboxdrv (proposed), if anyone is interested in this

joolswills, 2016-01-04:
I have made some improvements to the xboxdrv configuration menu. It uses a newer driver now, and won't overwrite your own config if you make one in /etc/rc.local

Happy to receive pull requests with further improvements though - for example generating the config based on the user choosing the number of controllers etc, and other settings.

joolswills, 2016-03-26:
Closing due to no feedback.

closed 2016-03-26 by joolswills

lawrencejd1978, 2020-04-09:
Hello everyone, I'm using pine a64 and installing retroarch, everything smooth but when i connect xbox 360 controller its keep blinking even i installed the driver using this command sudo apt-get install xboxdrv

can anyone help me out?

cmitu, 2020-04-09:
@lawrencejd1978 This is the RetroPie issue tracker - this is not the proper place to ask for support for Pine64 or RetroArch. Try the Pine64 forums.