RetroPie/RetroPie-Setup #830: Joysticks too sensitive

Alan-lubnow, 2015-05-11:
Anyone have issues with joysticks being too sensitive in Emulation Station? I built a PiCade and used the built in control configure in ES. It is wired properly and I was able to configure properly, but the movements are random, yet in the correct direction. Any known fixes?

joolswills, 2015-05-12:
using the picade usb board ? not seen this on mine, so it might be an issue with your board or picade specific.

NeoID, 2015-05-26:
Try this. I had the same problem with my THT controller interface:
http://blog.petrockblock.com/forums/topic/autofire-on-the-axis/#post-87758

joolswills, 2015-05-31:
Thanks - yeh it is likely to be that issue. @Alan-lubnow please test and let us know. thanks.

closed 2015-06-19 by joolswills