RetroPie/RetroPie-Setup #832: Possible Issue with "#" in ROM Filenames?

garrettmiller, 2015-05-14:
Hi all,

I think I might have a slight issue with certain GBA roms. Under 3.0b2, I was finding that roughly 50% of my roms were segfaulting, and roughly 50% were working. I think I’ve narrowed it down to a point where I found that the ones segfaulting were ones that contained a “#” in the filename. Output of runcommand.log is as follows:

```/opt/retropie/supplementary/runcommand/runcommand.sh: line 671: 2333 Segmentation fault /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-gpsp/gpsp_libretro.so --config /opt/retropie/configs/gba/retroarch.cfg "/home/pi/RetroPie/roms/gba/Advanced Wars # GBA.GBA" --appendconfig /tmp/retroarch.cfg```

The strange thing is, roms that worked under 2.6 don’t work under 3.0b2 – does it have anything to do with the “#” in the rom name? When I used the “rename” utility to rename these, it seemed to work. Behavior seems to be the same under lr-gpsp and gpsp.

Also posted about here:  http://blog.petrockblock.com/forums/topic/issue-with-gba-emulation-possible-filename-issue/

garrettmiller, 2015-05-17:
It may have to do with a difference of the ROM argument being passed in by "$2" instead of "$@" now?

joolswills, 2015-08-13:
It's a retroarch issue that needs debugging - Ill open a ticket at retroarch.

joolswills, 2015-08-13:
https://github.com/libretro/RetroArch/issues/2038

joolswills, 2015-08-13:
Actually it looks like it might be specific to gpsp. Not sure why they worked under 2.6 as I tried with an older libretro gpsp code and it still segfaulted, but it's reproducible from commandline, so it's not related to the way we launch stuff.

joolswills, 2015-08-13:
I have tested gpsp, and it works fine with roms with # in the filename - it's just the libretro core.

garrettmiller, 2015-08-13:
Cool - thanks for looking into it.  It was causing me some weirdness (my GBA romset had a number of #'s in filenames as it came) so that's what I was able to run down.  I wonder what's different about the libretro core vs. regular gpsp?  I had tried that as a debugging step myself but figured it might have been the way we were launching since regular gpsp didn't reproduce this error.

Thanks!

joolswills, 2015-10-15:
This is confirmed as a bug upstream, but it is being considered minor. in the meantime, people will just have to rename their romsets if they contain a #

garrettmiller, 2015-10-15:
Good to hear.  Thanks for the info!

garrettmiller, 2016-04-13:
As of 4/13/2016, this bug still exists.

joolswills, 2016-04-13:
You can follow the upstream ticket https://github.com/libretro/RetroArch/issues/2038

joolswills, 2016-10-22:
This is now resolved upstream - will close this once retropie binaries are updated with the fix.

joolswills, 2016-10-22:
RetroArch binaries have been updated.

closed 2016-10-22 by joolswills