RetroPie/RetroPie-Setup #834: Need support for Ipega PG-9025 controller

urtzai, 2015-05-18:
I'm trying to config my Ipega PG-9025 with no succeed... I'll be very appreciated to have some kind of setup, guidance...

Thanks

sur0x, 2015-05-20:
Im looking forward for this as well.

Did you tried this: 

http://askubuntu.com/questions/608386/using-ipega-bluetooth-gamepad-with-steam-linux

urtzai, 2015-05-20:
Nop! But thanks! I will. How do you pair your device with raspberry? I'm still stuck in that point...

Thanks for your info, I'll let you know if I achieve something.

urtzai, 2015-05-27:
I've created a question in stackoverflow, perhaps i'll get more clues... http://stackoverflow.com/questions/30481198/using-ipega-bluetooth-gamepad-with-retropie

urtzai, 2015-06-03:
Finally I found a solution: http://stackoverflow.com/questions/30481198/using-ipega-bluetooth-gamepad-with-retropie/30613619#30613619

I hope it will be usefull

closed 2015-06-20 by joolswills

wingsofsarrow, 2017-07-06:
do u guys know if you can use two ipega controllers at once for two player games

Zarroc1991, 2017-07-09:
I have the same question. How do I use 2 ipega controllers

platon-actual, 2018-11-23:
> Finally I found a solution: http://stackoverflow.com/questions/30481198/using-ipega-bluetooth-gamepad-with-retropie/30613619#30613619
> 
> I hope it will be usefull

Hello, sorry for reopening an old thread. Didn't work for me, as it looks, bluez-tools aren't installed on the latest Retropie release.
Still struggling with this.