RetroPie/RetroPie-Setup #836: Uae4Arm - alpha (retropie script attached) - Enhancement

amigagamer, 2015-05-19:
Uae4arm , based on UAE4all2 is the latest amiga emulator. Written by tomb for pandora who is the author of uae4all2  Its in ALPHA release stage at the moment, but brings a number of benefts even at this early stage: to name a few.

JIT compilation
Improved 68000 compatibility
68040 and 68881 emulation.
Command line switches.

Thanks to Chips@rpi forums we already have a working hardcoded Dispmanx version.

I have made a buildscript (which is also alpha release) based on the uae4all2 script released by buzz
https://github.com/amigagamer/uae4arm-retropie/blob/master/uae4arm.sh

Please note this is an ALPHA emulator release and an ALPHA buildscript. The emulator itself shows some significant improvement in 68000/68020 emulation, but the JIT code/fastest mode is very new and limited testing- expect bugs.
 Ive aimed to make this integrate with retropie/uae4all2 but im conscious uae4all2 is using a custom guichan.  This buildscript installs reference guichan081 and headers, from the raspbian repo so may cause incompatibilities with the preinstalled UAE4all2  ... Im using module_menus"4+" to flag this as experimental.

TLDR - use at your own risk!

@ retropie team, please feel free to copy/modify/change this script if you find it useful.

Further reading/information on this emulator is at 
http://boards.openpandora.org/topic/18067-alpha-uae4arm/
http://eab.abime.net/showthread.php?t=78037
https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=110488

joolswills, 2015-05-19:
guichan won't clash, as our uae4all builds and links against it statically. That was needed as it required patching to avoid some bugs. Do you know if this is fixed in the code above then as the debian guichan was not working correctly with uae4all.

joolswills, 2015-05-19:
thanks for the script.

amigagamer, 2015-05-19:
jools, the menu system is a complete rewrite (more of a winuae style) it plays much better with raspberry pi with reference guichan.  IVe set up HDF images and toggled lots of options without running into problems yet

another thing of note, we want to force this to run as "FBCON" not "DISPMANX" on your version of sdl as the dispmanx init and rendering is handled by uae4arm directly.  running both with dispmanx causes problem initialising the screen

joolswills, 2015-05-19:
yeh, retropie-setup will need some additional code for that. I would like to update the uae4all2 to include chips' code, but it never made it to github, and so it requires manual merging of changes etc (Ideally into the existing repository).

amigagamer, 2015-05-19:
I've been testing uae4arm with command line options. loading configs at start works well for example "uae4arm -f conf/a1200.uae"  This opens up the possibility of providing reference config files eg a500exact a1200exact a1200turbo and loading them from ES scripts.  May look at this for future script revision (i dont want to overcomplicate things right now)

I should have mentioned in original post due to cortex/neon optimisations this wont work on RPI1, its 2 only (will edit)

re uae4all2 , chips version has been on his github for a few weeks now

amigagamer, 2015-05-30:
@joolswills 
Learning github here slowly but surely.  Ive put this script into a "pull request" #846  ,hopefully ive got it right... Hope thats ok!

amigagamer, 2015-05-31:
This is now in retropie, with big thanks to @joolswills for the guidance.  
Nice timing for "International Amiga Day" :)

closed 2015-05-31 by amigagamer