RetroPie/RetroPie-Setup #838: Genesis plus gx not changing video modes properly

CharlesBukowski, 2015-05-21:
I'm having an issue with Genesis plus gx. I get a black screen with the game audio still playing if I add the following lines to "videomodes.cfg" in /opt/retropie/configs/all:

lr-genesis-plus-gx="CEA-16"
lr-genesis-plus-gx_render="output"

So I think the problem is with Genesis plus GX not switching video modes properly. This wasn't an issue until I reinstalled using the setup script a couple days ago (may 19).

joolswills, 2015-05-22:
the screenmode changing is not directly linked to the emulator - it is done before the emulator is launched. note for libretro emulators, you probably don't need to change video mode - unless you want a mode with a different refresh rate say for a pal game.

that file is not designed to be edited directly btw, but used via the pre-launch menu.

I can't help without more details though. But perhaps you can tell me why you are switching video mode, and whether other modes worked.  also telling the emulator to render at 1080p, may end up with shaders being slow.

CharlesBukowski, 2015-05-22:
Well, I want to completely avoid all scaling artifacts, and the only way to do this is by setting the render resolution to the output resolution, and setting output resolution to native. I also want to avoid the input lag that comes from scaling. By setting my video render to 1080p and render resolution to match, I'm avoiding scaling. I'm on a raspberry pi 2 so this has not affected performance. I don't care about shaders since I'm using an overlay to get scanlines. 

I don't notice the problem with any other emulators, just gen plus gx. I did alter the video mode via the launch menu. When I do this it causes gen plus gx to display a black screen with the game audio playing. I can only get the picture back by deleting the relevant lines in videomodes.cfg. I

I can confirm that other video modes do not work, either. The only thing that works is deleting the video settings, restoring them to their defaults, and then restarting the computer. 

I didn't have this problem until a reinstall a couple days ago (may 19).

joolswills, 2015-05-23:
Usually the pi will default to the native video resolution of your monitor - you shouldn't need to "switch" modes at all.

the pi has hardware scaling and retroarch can use it (hence being able to render at a lower resolution and then hardware scaled to the native res) - it should not affect input. there will technically be scaling artifacts if rendering to 1080p too (depending on config -  see http://blog.petrockblock.com/forums/topic/list-of-recommended-resolutions-for-each-system/)

I'll see if i can reproduce the problem - but no-one else has reported it so far, so it might be something on your system - have you tried increasing the amount of memory allocated for the gpu (in /boot/config.txt) ?

CharlesBukowski, 2015-05-26:
I started with a freshly initialized SD card and reinstalled everything using the script. Genesis plus gx is still exhibiting this problem. If I change my render resolution to "use video output" this causes a black screen although the game audio can still be heard.

CharlesBukowski, 2015-05-26:
I've tried fresh installs using 2 different SD cards and I am still experiencing this problem. My GPU memory is at 256.

CharlesBukowski, 2015-05-28:
I got gen plus gx to launch by selecting "cancel" from the launch menu. This causes the game to load normally. Choosing "launch" causes the game to load with a black screen.

joolswills, 2015-05-31:
strange since those should actually be identical in functionality (same code path).

You didn't respond to my point regarding not needing to switch screen mode - why not leave it at the default mode ? Are you saying not changing the video mode, but just changing the render resolution also gives you problems ?

CharlesBukowski, 2015-05-31:
I leave the video mode at the default and set retroarch render resolution to "use video output"

joolswills, 2015-06-19:
I've tried this, and it works fine on my system - sorry but unable to reproduce - no other users have reported this problem either. If you can get some more information or others users to reproduce we can reopen. thanks.

closed 2015-06-19 by joolswills