RetroPie/RetroPie-Setup #839: Overlay stability issue w/ NES and SNES

CharlesBukowski, 2015-05-21:
If I set "input_overlay" in retroarch.cfg, NES and SNES games will fail to load and RA crashes back to emulationstation. I've noticed this with the Nestopia and Snes9x-next cores. I've double and triple checked that the path is correct. 

I did not have this issue until a couple days ago, when I started with a fresh SD card and installed everything using the setup script.

joolswills, 2015-05-22:
please can you converse with other users on the forum first and verify if anyone else has the problem.

CharlesBukowski, 2015-05-22:
I've posted a topic on this to see if anyone replies. I also talked to hunterk of libretro, and he said there was a problem with overlay stability a couple weeks ago but that it should be fixed in the latest versions. 

I only notice the issue with Nestopia and SNES9x-next. 

Is there a way to update individual cores?

joolswills, 2015-05-23:
you can update individual cores from the retropie-setup menu - option 5 - you can build the very latest retroarch from source to see if it helps for example. you can also update from commandline using the retropie-packages.sh command.

CharlesBukowski, 2015-05-26:
I'm still getting this issue. How do I completely remove an emulator so that it can be reinstalled fresh? 

Thanks.

joolswills, 2015-05-26:
If you reinstalled from menu 5 - then it is installed from fresh pretty much. Sounds like there could be an upstream issue. I'll have to test it myself and try and reproduce.

CharlesBukowski, 2015-05-26:
I keep getting a segmentation fault in my error log. I'm going to try initializing my sd card and starting fresh to see if that solves it.

CharlesBukowski, 2015-05-26:
I reinstalled from a freshly initialized SD card and I'm still having issues with overlay stability with nestopia and snes9x-next. Starting either emulator with overlays enabled causes them to crash back to emulation station immediately upon launching a game. Overlays can be loaded within the emulator after it has been launched.

CharlesBukowski, 2015-05-26:
I'm also getting this problem with Genesis plus gx.
Haven't tested any other emulators

CharlesBukowski, 2015-05-26:
here's the error I'm getting in my reuncommand.log:

/opt/retropie/supplementary/runcommand/runcommand.sh: line 671:  5579 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-nestopia/nestopia_libretro.so --config /opt/retropie/configs/nes/retroarch.cfg "/home/pi/RetroPie/roms/nes/Batman - The Video Game.nes" --appendconfig /tmp/retroarch.cfg

The line that this appears to relate to is "input_overlay = path/to/file"

CharlesBukowski, 2015-05-28:
Here is another user reporting the same issue: 
http://blog.petrockblock.com/forums/topic/overlays-makes-games-crash/#post-98608

joolswills, 2015-08-02:
is this still happening ? A lot has changed with retroarch since this ticket and it may well be fixed now.

joolswills, 2015-08-02:
(Please try with 3.0 rc1)

joolswills, 2015-08-10:
Closing due to no feedback. Not heard reports about this with recent images, so suspect it was a retroarch issue that got fixed.

closed 2015-08-10 by joolswills