RetroPie/RetroPie-Setup #843: Conflict with retroarch controller files

biscuits99, 2015-05-24:
I think there is an issue with these 2 files:

/opt/retropie/emulators/retroarch/configs/Retrolink_USB_SEGA_Saturn_Classic_Controller.cfg
and
/opt/retropie/emulators/retroarch/configs/USB_Gamepad_.cfg

They both have the line:
input_device = "USB Gamepad "

For example, this thread:
http://blog.petrockblock.com/forums/topic/save-load-not-working/

amigagamer, 2015-05-26:
I made a post similar/possibly relating to this on the forums a while back, ill try to find it.

The conflict seems to be caused by all the retrolink pads use the same firmware/device IDs.  So the SNES/Saturn/Genesis/generic all identify themselves as teh same controller.  Think this is an inherited problem from retroarch.   Note the vendor and product IDs on the following 2 scripts - they match

https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/DragonRise_Inc._Gamepad.cfg

https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Retrolink_USB_SEGA_Saturn_Classic_Controller.cfg

joolswills, 2015-06-20:
we no longer install the auto configs by default so this issue should be better now. ES now handles setting up of retroarch controller config.

closed 2015-06-20 by joolswills