RetroPie/RetroPie-Setup #844: Joystick navigation is too sensitive in menu

NeoID, 2015-05-25:
I'm running EmulationStation my my Rpi2 and hooked up my joystick and buttons using this interface controller: http://www.jammaboards.com/store/pc-mame-ps3-to-arcade-controller-usb-interface-pcb-kit-pc2jamma-usb.html

I'm able to map the keys in the menu, but when trying to navigate it's way too "sensitive" when going in any direction. If I press down, it jumps 5-10 elements down. Is there a way to delay the input or only read input on key up event?

I've also tried to use RetroPie-Setup to configure my joystick, but once it asks me to press in any direction it triggers all the following input questions. In other words pressing in any direction triggers the key down event too many times in order to be able to navigate properly.

NeoID, 2015-05-26:
Confirmed bug.
This solves the issue: http://blog.petrockblock.com/forums/topic/autofire-on-the-axis/#post-87758

closed 2015-05-26 by NeoID

joolswills, 2015-05-26:
I should have replied earlier sorry - I suspected that, but didn't realise your controller board was affected. Anyway - glad you have the solution. Hopefully this fix will make the upstream kernel - I commented a bit on that before on the pi firmware bugtracker - not sure if it got submitted upstream though. I'll check.