RetroPie/RetroPie-Setup #845: adding wonderswan [beetle-wswan-libretro]

InsecureSpike, 2015-05-30:
please could we have beetle-wswan-libretro for wonder swan support, as posted here...

http://blog.petrockblock.com/forums/topic/wonderswan-coleco-emulation-trouble/

thank you

HerbFargus, 2015-05-31:
Well, I sorta got two out of three. I wasn't able to compile the game and watch emulator yet- I don't know enough coding for that one.

WONDERSWAN:
https://github.com/petrockblog/RetroPie-Setup/pull/849

NEO GEO POCKET:
https://github.com/petrockblog/RetroPie-Setup/pull/850

InsecureSpike, 2015-05-31:
awesome! many thanks, i have no idea if Game & Watch works i'm in the same position as you.
just thought someone could give it a go,
thanks

joolswills, 2015-05-31:
I built game and watch, but I can't get it working correctly - it shows a photo of the game with the led on, and some buttons highlight etc, but it seems to freeze moments later.

here's the build script if you want to try it 

```
# (c) Copyright 2012-2015  Florian Müller (contact@petrockblock.com)
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/petrockblog/RetroPie-Setup/master/LICENSE.md.
#

rp_module_id="lr-gw"
rp_module_desc="Game and Watch simulator"
rp_module_menus="4+"

function sources_lr-gw() {
    gitPullOrClone "$md_build" https://github.com/libretro/gw-libretro.git
}

function build_lr-gw() {
    make -f Makefile.libretro clean
    make -f Makefile.libretro
    md_ret_require="$md_build/gw_libretro.so"
}

function install_lr-gw() {
    md_ret_files=(
        'gw_libretro.so'
        'LICENSE'
        'README.md'
    )
}

function configure_lr-gw() {
    mkRomDir "gameandwatch"
    ensureSystemretroconfig "gameandwatch"

    addSystem 1 "$md_id" "gameandwatch" "$md_inst/gw_libretro.so" "Game and Watch" ".mgw"
}
```

HerbFargus, 2015-05-31:
I tested it out with donkey kong and it seemed to work fine. Once you die you just have to select the game A or B button (left or right trigger) to restart the game otherwise it will seem frozen. which game were you testing?

joolswills, 2015-05-31:
parachute - but thanks for the info. I did get some crashes too, so it has bugs, but if it does work mostly, ill add it under the experimental. I was testing with a keyboard, and I probably had the mapping wrong. thanks for the feedback.

joolswills, 2015-06-01:
Closing as all added. thanks.

closed 2015-06-01 by joolswills