RetroPie/RetroPie-Setup #853: new emulationstation and custom retropie menu

joolswills, 2015-06-01:
@petrockblog 

I am just testing the new emulationstation code, but there are issues with the retropiemenu - the new unstable branch uses sqlite for the gameslists, so our code would need to be changed to get the correct labels etc for the menu.

I'm not sure if the database stuff is finalised/complete and we may run into problems running off the unstable branch - perhaps it might be worth considering rebasing your changes on top of the master branch for now (Would it require major changes or ?)

Does your current code handle keyboard input - I don't think it does right ? If not - it should perhaps not request the full choice of key input if a keyboard is used, or the code could be updated to set up the keyboard correctly in the config which would be useful for things like xarcade sticks, and control boards like the one on the picade etc.

joolswills, 2015-06-01:
Problems noticed currently:

New emulationstation startup time (with no roms), is very slow. - i guess the sqlite code needs some work.

Had just a keyboard plugged in. Asked me to configure far more controls than needed - should handle this case.

Quit and restarted. the second time it picked up the retropiemenu xml, and asked to import it (It should do this on the first load really). The yes/no requester doesn't disappear after choosing an option, so it seems crashed whilst it imported the retropie menu. it took about 10 second+ to import just these few items, so something isn't running as it should.

What do you want to do - do you have any free time to do any further work on this - I'm not sure this is ready for the next beta though which I would like to get out asap - so perhaps we can keep the old emulationstation for now until we both have time to improve on this.

petrockblog, 2015-06-01:
I take a look into this right now. I have cherry-picked the changes to the master branch. Now I am looking into the keyboard control issues.

The list of controls to be configured is the same as retroarch-joyconfig. I agree that his could be improved to faster skip the unneeded controls, but I do not see at at top priority now. BTW, you can hold a key to skip an unneeded key.

I will write here later as soon as I have new information..

joolswills, 2015-06-01:
Thanks! :) I really like the new system overall - so having it work for keyboard, and with the older emulationstation would certainly make it good to go for now.

I was just reading an article about sqlite3 performance on the pi - http://spin.atomicobject.com/2013/11/14/sqlite-raspberry-pi/ - perhaps we can look at improving that in the future on the unstable (I'd be quite interested in having a play with the code). It does seem that emulationstation development is not doing too well - although there has been some activity, there is a lack of overall organisation I think - with users pull requests not getting any attention. We might end up having to maintain this ourselves.

HerbFargus, 2015-06-01:
if I remember correctly when I was testing the first build- once I had the controls configured that I needed I pressed F4 and I'm pretty sure the first few inputs were saved in the configurations, that way I didn't have to hold the button down for every control that I didn't need. Is it still that same way?- because I think it would be incredibly convenient to press one button to exit  and save once you configured all the buttons you need so you don't have to go through the rest of the unnecessary button presses.

petrockblog, 2015-06-01:
All right, I see that I have not tested the current implementation with a keyboard only. It seems that EmulationStation is working as intended. However, I need to adapt the supplementary scripts for the keyboard-only support in https://github.com/petrockblog/RetroPie-Setup/tree/master/supplementary/moduledata/supplementary/emulationstation. I won't finish that today, but I plan to do that tomorrow!

petrockblog, 2015-06-01:
Just that you know, the scripts need to be adapted such that EmulationStation and RetroArch configurations are set given this input:
```xml
<?xml version="1.0"?>
<inputList>
        <inputConfig type="keyboard" deviceName="Keyboard" deviceGUID="-1">
                <input name="a" type="key" id="97" value="1" />
                <input name="b" type="key" id="115" value="1" />
                <input name="down" type="key" id="1073741905" value="1" />
                <input name="left" type="key" id="1073741904" value="1" />
                <input name="right" type="key" id="1073741903" value="1" />
                <input name="select" type="key" id="32" value="1" />
                <input name="start" type="key" id="13" value="1" />
                <input name="up" type="key" id="1073741906" value="1" />
                <input name="x" type="key" id="122" value="1" />
                <input name="y" type="key" id="120" value="1" />
        </inputConfig>
</inputList>
```

joolswills, 2015-06-01:
thanks. so there will be a need to convert between the id's and keypresses I guess - can probably have an associative array with the data from https://wiki.libsdl.org/SDLKeycodeLookup or so ?

I'm happy to contribute too as soon as I have some more free time. cheers.

petrockblog, 2015-06-01:
Thanks for that link!

joolswills, 2015-06-05:
Did you make any progress with the additional bash script to handle keyboard input for retroarch ? If not and you do not have any free time I may look at it myself.

I had a quick look the the current scripts and I think they could do with a few cosmetic tweaks. Some of the quoting styles and naming does not seem to match existing styles and I think we could simplify / shorten function names (without the worry of any collisions).

joolswills, 2015-06-05:
I also think they should be moved under the scriptmodules/supplementary directory to keep them together with es - I think we should probably do the same for the runcommand code too - we can adapt the module scanning to avoid them being picked up etc by that (if it does currently). Thoughts ?

petrockblog, 2015-06-06:
Hi, unfortunately, I did not find any more time until now. I hope to further work on that tomorrow. However, I would also be happy if you would have a look at that. 
I am looking forward to any cosmetic tweaks or code clean ups!

petrockblog, 2015-06-07:
Are you alreeds working on that? If not, I would continue with that now. Just want to avoid any later merging issues ...

joolswills, 2015-06-07:
Not done anything yet - Please carry on.

petrockblog, 2015-06-07:
Right in the middle of it.

joolswills, 2015-06-07:
Looking forward to trying it out :) Cheers

petrockblog, 2015-06-08:
All right, I have just pushed (921bfde1c111a84dea8d32193807c6c121c9ebf0) the keyboard support for the controller configuration scripts. Curious about feedback.

joolswills, 2015-06-08:
thanks ! will give it a go.

joolswills, 2015-06-09:
So it all seems good so far on my system. I will sort out some new beta images shortly.

There are a few usability issues I think we should address:

* Currently when running the scripts after configuring inputs in emulationstation it can look as though it has frozen after pressing "ok". A window saying that controls are being configured would be better - this requires some emulationstation changes of course.

* It would be useful to perhaps hide some of the inputs for keyboard users, and/or allow quicker skipping - currently can take a little while to skip each unneeded control. This is not a major problem though as the idea is this is only done once etc.

I am currently looking into allowing joystick controls in our retropie-setup menus - which would make some configuration even easier. The tool joy2key partly helps, but it doesn't allow multiple character presses such as arrow keys - however I've modified it so it does do this, so I'll probably stick this up on github with some changes at some point. It's not a massive problem for users to use a cheap bluetooth keyboard though etc for setup.

petrockblog, 2015-06-09:
> Currently when running the scripts after configuring inputs in emulationstation it can look as though it has frozen after pressing "ok". A window saying that controls are being configured would be better - this requires some emulationstation changes of course.

I totally agree to that. I did not dig that deep into EmulationStation so far so quickly add that, though.

> It would be useful to perhaps hide some of the inputs for keyboard users, and/or allow quicker skipping - currently can take a little while to skip each unneeded control. This is not a major problem though as the idea is this is only done once etc.

If someone has a good idea to quickly skip the process, we could add that. But, yes, the control configuration should only be done once or twice in practice.

Looking forward to the joy2key addition.

BTW: I have added another default splashscreen with the new logo. Could you please add that into the new image?

joolswills, 2015-06-09:
Will do - I'm going to rebuild everything for rpi1/rpi2 then I'll regenerate the images and upload them. Should have this ready tomorrow - I'll put together a changelog too. Cheers.

taalas, 2015-06-11:
Joy2key sounds like an excellent idea for the dialog-based setup bash scripts!

HerbFargus, 2015-06-12:
The new controller module is great, I also agree it could use a notification screen while it configures, and I second the joy2key idea- it could greatly improve things.

As far as ideas for skipping, I was thinking either a keyboard key thats rarely used could be pressed to save the changes once the needed keys are pressed (which would save the changes and cut out the skipping each key one by one altogether) or maybe a key combo could work like pressing a+b to skip each unneeded button press which would eliminate the need for a keyboard to skip.

KrunchMuffin, 2015-06-12:
I have one of those BT kb/mouse pad units and installed B3. I am correct in assuming, according to this thread, there is no way to skip all that to get to the main config? The pad input was not working so I couldn't just click OK. Is there an easy to change to the Dev branch?

joolswills, 2015-06-12:
you will need to pair your device first manually from the console - this was required on previous versions too - unless I'm missing something ?

http://elinux.org/RPi_Bluetooth_keyboard_setup

KrunchMuffin, 2015-06-12:
even if the keyboard works?

joolswills, 2015-06-12:
the bluetooth keyboard works without pairing it ? Perhaps you can start a thread on the forum with more information about your setup/hardware.

KrunchMuffin, 2015-06-12:
ya, never have had to pair it like that with any images I have used.

joolswills, 2015-06-12:
Well, usually bluetooth devices need pairing - nothing has changed in that respect. Apart from additional controls being configured within emulationstation, nothing else has changed - so it should behave the same. Without details of your setup etc I won't be able to help further - but try first on the forum, and if there is a bug, we can open a new issue.

KrunchMuffin, 2015-06-12:
I screwed up, it isn't BT, it's just the regular wireless, can't remember the term for it right now.

joolswills, 2015-07-16:
Closing this as the main issues have been resolved. I will continue any usability changes over at our es fork.

closed 2015-07-16 by joolswills