RetroPie/RetroPie-Setup #854: Adding: Super Mario War

InsecureSpike, 2015-06-02:
could this be added to the experimental menu please.

http://blog.petrockblock.com/forums/topic/super-mario-war/

thanks!

epicTCK, 2015-07-16:
Why under experiamental menu? why not under ports?

joolswills, 2015-07-16:
Because new stuff often stays in experimental until fully tested etc.

HerbFargus, 2015-07-26:
It has been added: https://github.com/RetroPie/RetroPie-Setup/pull/917 
I think this can now be closed.

joolswills, 2015-07-26:
Will close - btw we have a fair amount of "experimental" cores etc, I would be interested in user feedback and of course your testing so as to move them to stable.

I'm also thinking about changing our flags, so we have "core" "base emulators" "extras" and "experimental". meaning we could ship with less stuff by default but have it easily installable. would solve the issue that many people dont consider mac emulation a basic thing.

Ill think more and perhaps come up with something for discussion of course.

closed 2015-07-26 by joolswills

HerbFargus, 2015-07-26:
Sounds good to me. I'll create a spreadsheet or something for all the experimental cores and maybe open it up to the forum to contribute compatibility as well like I did with the mame lists.

joolswills, 2015-07-26:
that would be excellent! btw - I really appreciate your recent work! I can be somewhat ocd when it comes to code, but it's all appreciated - thanks! :)

HerbFargus, 2015-07-26:
No problem,  I'm happy to help :) its important to have consistency in the code and having that standard helps me learn. Now to hash out one last module for miroof's virtual gamepad...

petrockblog, 2015-07-26:
@joolswills I like your idea about changing the flags. It would be a pity, though, if the large set of supported emulators would be hidden too deep and people would not find them.
Maybe this would be a good opportunity to think about a one-time configuration dialog for the first boot of a RetroPie image. Here, the root file system would be expanded and the user could select the modules that he is initially interested in.

joolswills, 2015-07-26:
yeh - just ideas now anyway - but that would def simplify it. perhaps something for 3.1 or 4.0 :)