RetroPie/RetroPie-Setup #855: Looking to remove unwanted controllers, and forcing others to be player 1.

lewiswake, 2015-06-08:
So, I originally mapped a wired xbox controller as player 1 (js0), and it seems I've managed to map the same controller as player 2 (js1).
I've just got a bluetooth dongle, and have sucessfully mapped two playstation 3 controllers as player 3 and 4 (js2, and js3 respectively).

I've browsed several forums and guides, and I'm looking to remove the xbox controller as player 1, and force the playstation 3 controllers up to player 1 and 2 in emulationstation?

Any advice? I'm running the latest version of RetroPie on a Pi 2.

If this is not the correct place to post this request, I apologise and can you please direct me to the correct place to post this.

joolswills, 2015-06-08:
You may get more help asking on the http://blog.petrockblock.com/forums/forum/retropie-project-forum/ (Where there are other users who may have a similar setup)

lewiswake, 2015-06-09:
Thank you, feel free to close this post if possible.

closed 2015-06-19 by joolswills