RetroPie/RetroPie-Setup #858: simplified static canela new logo v2.0 label:enhancement

InsecureSpike, 2015-06-11:
hi Florian, the theme has been updated with the new logo, (and new text)
i've remade some of the banners and added 5 new ones.

the zip contains the theme and a boot screen too

https://www.mediafire.com/?46ermnrvcp221fc

robertybob, 2015-06-11:
Great work!

robertybob, 2015-06-11:
Any chance of this making the next beta @joolswills?

joolswills, 2015-06-11:
We missed the last one, but I'll add it before the next release (which may not be a beta depending how this one goes). thanks.

robertybob, 2015-06-21:
Thanks @joolswillis , I presume it'll be Beta 5 (or the equivalent) ?

joolswills, 2015-08-13:
It didn't make the last release, but it is included now if you update retropie-setup, you can install it from the supplementary menu.

closed 2015-08-13 by joolswills