RetroPie/RetroPie-Setup #860: BETA 3: Root password

petrockblog, 2015-06-12:
What should we do with the default root password settings of the image? See https://twitter.com/max521/status/609142986224611328. Should we just keep it unset or also set it to "raspberry"?

joolswills, 2015-06-12:
I intentionally set it back to like Raspbian - Never noticed it was enabled before, but I think its safer to have to disabled - it's easy for users to enable by just setting a password (as with raspbian)

joolswills, 2015-06-12:
Users can be pointed to http://elinux.org/R-Pi_Troubleshooting#I_don.27t_know_the_root_password

joolswills, 2015-06-12:
Some additional information - 

https://www.raspberrypi.org/documentation/linux/usage/root.md

>You can also run a superuser shell by using sudo su. When running commands as a superuser there is nothing to protect against mistakes that could damage the system. It is like disabling the safety guards on a machine. It makes it easier to access the insides, but the risk of damage is far greater. It is recommended that you only run commands as the superuser when required and to exit a superuser shell when it is no longer needed.

mariano, 2015-06-12:
@petrockblog this is a non-issue for me. This is basic Debian knowledge, a password shoud not be set for root. The default user has sudo access, so a password can be set by simply:

```bash
$ sudo passwd root
```

I'd add that to the FAQ, and close this issue.

HerbFargus, 2015-06-12:
I've added it to the FAQ on the Wiki: https://github.com/petrockblog/RetroPie-Setup/wiki/Root-Password
I think we can consider this closed.

joolswills, 2015-06-12:
thanks all.

closed 2015-06-12 by joolswills