RetroPie/RetroPie-Setup #863: RGUI Freezes in RetroPie 3 beta 3

biscuits99, 2015-06-12:
When in-game, if you bring up RGUI using the hotkey combo of select and X, it starts, but freezes on title screen. Was fine on RetroPie 3 beta 2, seems broken in beta 3.
I am using an iBuffallo USB joypad, but seems to happen on various others.

joolswills, 2015-06-12:
please can you try building a new retroarch from source in case we built it during a problem. If it still has an issue, we may need to trace back as to when the problem started and feed it back to retroarch

HerbFargus, 2015-06-12:
I tested it, and I don't think it is freezing- it just doesn't register joypad inputs. I opened the rgui with the hotkey combination and couldnt navigate with my joypad but as soon as I pressed enter on my keyboard the rgui menu was working fine.

HerbFargus, 2015-06-13:
From a fresh retropie 3 BETA 3 image attemping to install from source I get this error: 

Could not successfully build retroarch RetroArch (/home/pi/RetroPie-Setup/tmp/build/retroarch/retroarch not found)

This is my full error log:

http://pastebin.com/5aeE4S3d

joolswills, 2015-06-14:
Compiles fine for me currently - maybe you were just unlucky and built it when some changes had been made and some fixes were needed. RetroArch is in constant development.

HerbFargus, 2015-06-14:
its possible. perhaps I'm doing it the wrong way- I've been selecting the option in the setup script of update individual emulators from binary or source. I just tried it again and it gave me the same error. Is there a different method of compiling it?

joolswills, 2015-06-14:
aha. i was cross compiling on gcc 4.8 - looks as though there is an issue due to this https://gcc.gnu.org/bugzilla/show_bug.cgi?id=52624 - I'll put in a workaround for now, and feed it back to retroarch.

joolswills, 2015-06-14:
https://github.com/libretro/RetroArch/issues/1793

joolswills, 2015-06-14:
please try now.

HerbFargus, 2015-06-14:
you're brilliant- now it compiles - but..... the rgui still freezes.

joolswills, 2015-06-14:
np I will need to git bisect it to see when the problem started etc. Will hopefully find time for this in the next day or two.

joolswills, 2015-06-14:
bisecting the menu issue now.

joolswills, 2015-06-15:
It's quite difficult to pinpoint where the problem started due to various issues appearing during recent development (there isn't a specific point where it works then it stops working, there are various commits where certain functionality breaks ending up as we are now).

While I continue looking into this to send a report upstream, I will lock retroarch to an older stable version, and rebuild the binaries.

joolswills, 2015-06-15:
I have switched retroarch to an older revision - no binaries updated yet - but if you build from source, it should now work again.

HerbFargus, 2015-06-15:
after building from source I can verify that it is now working.

joolswills, 2015-06-16:
I have reported the issue upstream here - https://github.com/libretro/RetroArch/issues/1807 I will likely need to provide further information to debug this - depending if the main developer(s) have rpi hardware and can reproduce it.

joolswills, 2015-06-19:
guess we can close this now - as we have a workaround. At some point I should be able to remove it, once they solve this upstream.

closed 2015-06-19 by joolswills