RetroPie/RetroPie-Setup #864: InputStation writes D-Pad as _btn and not _axis

biscuits99, 2015-06-12:
When configuring my iBuffalo joypad in Emulation Station, the file it writes for RetroArch logs my D-Pad as ending in _btn instead of _axis.
The result is I cant move up/down/left/right in games. I resolved it by manually editing the created file in /opt/retropie/emulators/retroarch/configs/

For example:
input_up_btn = "-1"

instead of:
input_up_axis = "-1"

HerbFargus, 2015-06-12:
I can also verify that this is an issue as well as the solution provided by @biscuits99 
I replaced these 4 lines:
```
input_up_btn = "-1"
input_right_btn = "+0"
input_down_btn = "+1"
input_left_btn = "-0"
```
with these 4
```
input_up_axis = "-1"
input_right_axis = "+0"
input_down_axis = "+1"
input_left_axis = "-0"
```
and it solved the issue.

joolswills, 2015-06-13:
@petrockblog is it possible to tell whether we need a _btn or _axis setting for retroarch from the temporary generated config from ES ? Do you know what needs to be done to sort - It's fine for my gamepad as that uses _btn.

joolswills, 2015-06-13:
Actually, I'll recheck mine as I have the same/similar controller - http://www.amazon.co.uk/gp/product/B002B9XB0E (which is a lot cheaper than when I bought mine in April)

biscuits99, 2015-06-13:
@joolswills  It looks like the file es_temporaryinput.cfg will indicate if it is axis - at least it looks good on my iBuffalo. Each D-Pad direction is marked axis below.

`<inputList>
        <inputConfig type="joystick" deviceName="USB,2-axis 8-button gamepad  " deviceGUID="03000000830500006020000010010000">
                <input name="a" type="button" id="0" value="1" />
                <input name="b" type="button" id="1" value="1" />
                <input name="down" type="axis" id="1" value="1" />
                <input name="left" type="axis" id="0" value="-1" />
                <input name="leftbottom" type="button" id="4" value="1" />
                <input name="right" type="axis" id="0" value="1" />
                <input name="rightbottom" type="button" id="5" value="1" />
                <input name="select" type="button" id="6" value="1" />
                <input name="start" type="button" id="7" value="1" />
                <input name="up" type="axis" id="1" value="-1" />
                <input name="x" type="button" id="2" value="1" />
                <input name="y" type="button" id="3" value="1" />
        </inputConfig>`

joolswills, 2015-06-13:
ok cool - I shall debug/fix up the retroarch config generation script then. Cheers.

petrockblog, 2015-06-14:
Yes, I was not aware of that (too few gamepads tested ;-) ). All information that are needed for the script are available, e.g., at https://github.com/petrockblog/RetroPie-Setup/blob/master/supplementary/moduledata/supplementary/emulationstation/configscripts/retroarch.sh#L24.

joolswills, 2015-06-14:
Please try my branch with this hopefully fixed (if you get permission denied on any of the commands it's because something has become owned by root - you will have to sudo it - we need to fix up the updating of the script from retropie-setup as it does it from root rather than the pi user)

```
cd ~/RetroPie-Setup
git remote add jools https://github.com/joolswills/RetroPie-Setup.git
git fetch jools
git checkout jools/input_fixes -b input_fixes
sudo ./retropie_packages.sh emulationstation configure
```

HerbFargus, 2015-06-14:
I tried your branch but now it doesnt create retroarch configs at all. /opt/retropie/emulators/retroarch/configs is empty after setting up the controls in emulationstation. 
/home/pi/.emulationstation/es_input.cfg was made though and it looks like this:

```shell
<?xml version="1.0"?>
<inputList>
	<inputConfig type="joystick" deviceName="USB gamepad           " deviceGUID="030000001f08000001e4000010010000">
		<input name="a" type="button" id="1" value="1" />
		<input name="b" type="button" id="2" value="1" />
		<input name="down" type="axis" id="1" value="1" />
		<input name="left" type="axis" id="0" value="-1" />
		<input name="leftbottom" type="button" id="4" value="1" />
		<input name="right" type="axis" id="0" value="1" />
		<input name="rightbottom" type="button" id="5" value="1" />
		<input name="select" type="button" id="8" value="1" />
		<input name="start" type="button" id="9" value="1" />
		<input name="up" type="axis" id="1" value="-1" />
		<input name="x" type="button" id="0" value="1" />
		<input name="y" type="button" id="3" value="1" />
	</inputConfig>
</inputList>
```

joolswills, 2015-06-14:
and if you run 

```
/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
```

manually ?

The branch works fine here and correctly creates the retroarch config, so not sure what could be wrong.

HerbFargus, 2015-06-14:
ah yes i missed that step. Runs brilliantly mate :) 

just for reference this is my  /opt/retropie/emulators/retroarch/configs file that was created:

```shell
input_device = "USB gamepad           "
input_driver = "udev"
input_up_axis = "-1"
input_right_axis = "+0"
input_state_slot_increase_axis = "+0"
input_down_axis = "+1"
input_left_btn_axis = "-0"
input_state_slot_decrease_btn_axis = "-0"
input_a_btn = "1"
input_b_btn = "2"
input_reset_btn = "2"
input_x_btn = "0"
input_menu_toggle_btn = "0"
input_y_btn = "3"
input_l_btn = "4"
input_load_state_btn = "4"
input_r_btn = "5"
input_save_state_btn = "5"
input_start_btn = "9"
input_exit_emulator_btn = "9"
input_select_btn = "8"
input_enable_hotkey_btn = "8"
```

joolswills, 2015-06-14:
that script should have been run after re-entering emulationstation and doing a new config though - can you confirm that works too after doing "sudo ./retropie_packages.sh emulationstation configure"

HerbFargus, 2015-06-14:
it didn't run that script when i ran "sudo ./retropie_packages.sh emulationstation configure" it's possible added your branch wrong but yeah it isn't running it.

joolswills, 2015-06-14:
Sorry - some confusion here. sudo ./retropie_packages.sh emulationstation configure doesn't run it - that just installs the script and resets the emulationstation configuration so when you load emulationstation it will ask for control configuration. after control configuration the script above is run.

HerbFargus, 2015-06-15:
gotcha, yeah after i did the sudo ./retropie_packages.sh emulationstation configure thing and ran the configuration again through emulationstation it still didn't create the retroarch configs

joolswills, 2015-06-15:
odd, as all it does is run the script which worked when you ran it manually. You didn't run it as sudo or anything manually ? Please can you give me the output of

```
ls -la /opt/retropie/emulators/retroarch/configs
```

HerbFargus, 2015-06-15:
actually on further testing, I ran it again and rebooted and it did create it. my bad. I may have ran  the config from within the emulationstation start menu. Does that behave differently than the first boot configuration?

joolswills, 2015-06-15:
yeh - that's something else that needs sorting :)

HerbFargus, 2015-06-15:
ok that is good to know, so yes other than the in menu thing your patch works :)

joolswills, 2015-06-15:
now we have the retroarch configuration generated, there is no reason for us to have all the autoconfig files installed by default - do you agree ?

I could leave it as an option perhaps in supplementary perhaps but I'm not sure there is a use now unless someone wants to use retropie without es.

HerbFargus, 2015-06-15:
I agree- as we've seen with a lot of posts on the forum (http://blog.petrockblock.com/forums/topic/usb-snes-controller-d-pad-not-working-in-games/#post-97091) the autoconfigs have conflicts that muck things up. It may be useful to have them as a reference somewhere as you said but it would make things cleaner if they were not implemented by default.

joolswills, 2015-06-15:
I also think the joystick configuration should live in /opt/retropie/configs/all/joypads or something too - what do you think ?

joolswills, 2015-06-15:
thanks for the feedback btw. appreciated

HerbFargus, 2015-06-15:
I'm happy to help- as far as joypads are you saying to make a distinction from keyboard configs? Since its a new configuration set up I think that /opt/retropie/configs/all/joypads is a good idea

joolswills, 2015-06-15:
well, keyboard configs are still kept in the main retroarch.cfg - the current configs folder with the joypad configurations is pointed to by the ini option "joypad_autoconfig_dir" - and it only contains joypad configuration. If I have understood your question correctly.

My plan is:

Don't install retroarch joypad autoconfigs by default.
Move the folder location to /opt/retropie/configs/all/joypads
if the user does use the autoconfig module to install the many joypad config, it will place the files also in /opt/retropie/configs/all/joypads (easy access to remove/delete)

joolswills, 2015-06-15:
maybe the folder should be called "retroarch-joypads" to avoid confusion. to be honest, we should probably have a configs/retroarch folder for retroarch main configs rather than "all", but its a pita to move everything.

HerbFargus, 2015-06-15:
yeah I think that is sound judgement- it sounds good to me. And yeah it probably would make more sense to have a configs/retroarch folder as well but yeah i can see how it would be tedious to change everything up like that. I figure as long as the documentation is clear enough then people can figure it out and maybe down the road once the main coding for controller configuration is sorted out some of the config files can be streamlined more.

joolswills, 2015-06-15:
I assume the autoconfigs could be useful when using multiple controllers - This is something I have not looked into at all yet - I have just purchased a second controller so I can test when that arrives. I assume ideally we would be able to configure multiple controllers from emulationstation. Any ideas/feedback welcome.

HerbFargus, 2015-06-15:
I have two controllers that I'm working on testing right now with your patch. I think if the in menu configuration is sorted out to function as it does on first boot that should sort out multicontroller use yes?

joolswills, 2015-06-15:
yeh if it had changes that allow reconfiguration of primary controller (for es), and then additional ones etc.

HerbFargus, 2015-06-15:
There's a slight typo that causes an error i just noticed- 
```
input_left_btn_axis = "-0"
```
Should be
```
input_left_axis = "-0"
```
otherwise the left dpad doesn't work.

As far as multiple controllers, I have 4 controllers- and tested by plugging them into many different variations and found that with autoconfig files, the order of the player is based on where they are plugged in rather than the order in which the config file was created- it goes top to bottom left to right
 from the top left usb port.  

[image]

(as of right now because the in menu configuration in emulationstation doesn't configure retroarch controls I had to unplug all the controllers but one and reboot to configure the one controller and had to repeat the process for each different controller)

joolswills, 2015-06-15:
well spotted - fixed thanks. Somehow I missed it when search/replacing - There was another case too which is now sorted.

Thanks for the info regarding joysticks - good to know.

petrockblog, 2015-06-15:
> actually on further testing, I ran it again and rebooted and it did create it. my bad. I may have ran the config from within the emulationstation start menu. Does that behave differently than the first boot configuration?

Actually, it is _supposed_ to to the same. But there might be a bug, though.

> autoconfig plans: 
>  - Don't install retroarch joypad autoconfigs by default.
 - Move the folder location to /opt/retropie/configs/all/joypads
 - if the user does use the autoconfig module to install the many joypad config, it will place the files also in /opt/retropie/configs/all/joypads (easy access to remove/delete)

I second that. Generally, the autoconfig capability allows users to configure multiple controllers and hot-swap these. I also think that we could keep the retroarch joypad autoconfigs away per default so that they do not interfere with the users configuration from the very beginning.

HerbFargus, 2015-06-15:
actually on further testing, the configuration menu within emulationstation does work in creating retroarch configs just like the one that starts on first boot, i must have been doing something wrong when I was testing on the branch- so that's all good.

biscuits99, 2015-06-15:
Totally agree with the removal of autoconfigs - when users report issues with thesir controller, there are sometimes conflicting files in there confuses them. Most users just dont care and want the controller to work.
Doing this via ES will really help - perhaps subsequent controllers could be mapped using the retroarch script in the retropie menu in ES (sounds like this works now).

Can someone remind me what happens when
- User creates ES (and retroarch) controller config for the first time
- Then they reboot but have a different controller plugged in
- Does ES want to update its settings and also create a new retroarch config?

Ignore that last bit if I'm going off topic.

HerbFargus, 2015-06-15:
This is what I understand thus far- feel free to correct me if I'm wrong
 * It creates 2 new configuration files -maybe 4 if the N64 was incorporated (more will be added for other emulators in time) 
- emulationstation configurations: `/home/pi/.emulationstation/es_input.cfg`
- retroarch configs: `/opt/retropie/emulators/retroarch/configs/yourgamepad.cfg`

I thought that mupen64plus/n64 configs were also created- but i didn't remember seeing them on this last test- i may have to do more testing on that one. For reference though the files should be:

`/opt/retropie/configs/n64/InputAutoCfg.ini`

`/opt/retropie/emulators/mupen64plus/share/mupen64plus/InputAutoCfg.ini`

I was looking at your video on it floob and I'm not sure if the InputAutoCfg.ini controls were created since your gamepad is in there by default:
https://github.com/mupen64plus/mupen64plus-input-sdl/blob/master/data/InputAutoCfg.ini#L1584

 * if they reboot and have a different controller, it will only ask for reconfiguration if that new controller is plugged in by itself. If another controller that has already been configured is plugged in with the new controller it won't ask for new configs on boot, But you can press start while in emulationstation and choose configure input and configure your new gamepad from there- it will then add its configs to emulationstation `/home/pi/.emulationstation/es_input.cfg` and it will create retroarch configs here as well  `/opt/retropie/emulators/retroarch/configs/yourgamepad.cfg`

* and the above should have answered your third question as well if I read it correctly.

biscuits99, 2015-06-15:
@HerbFargus  Ah - yes, I think you may be right and that my controller settings were already there. I really should have checked before I ran ES.

That sounds good - I'll re-run this on the new beta when its released and clarify how this works for everyone with a vid. A clean retroarch config directory should help debug user controller issues as well I think.

joolswills, 2015-06-16:
Continuing from https://github.com/RetroPie/RetroPie-Setup/commit/a6bb217e2e1c116ba02e7f47449455b0544278c5#commitcomment-11697073

>So on initial configuration the keyboard works as it should but as soon as I exit out of a game its as if it just totally forgets that it is configured and no longer works in emulationstation or retroarch but the configurations are still there both in emulationstation and retroarch. Also when I configure it on its own through the start menu it doesn't seem to recognise anything has changed where configuring it on boot (while the keyboard is the only thing plugged in) at least works once until you exit out of a game. Its kinda weird, are you seeing the same things with your build?

The bug with ES "forgetting" keyboard configuration was due to me moving some code around. about to fix that.

joolswills, 2015-06-16:
if you update the script and then do

```
sudo ./retropie_packages.sh emulationstation configure
```

the keyboard issue should be sorted. please let me know. not able to reproduce any problems with retroarch but perhaps it was somehow related (Wasn't sure from your text).

HerbFargus, 2015-06-16:
Ok works brilliantly again and it remembers the keyboard configs but there seems to be an issue with the keyboard hotkeys  so that when I configure my gamepad after I've configured my keyboard the hotkeys no longer work (gamepad hotkeys still work though)- I'm just going to make sure it wasn't some of the configs I messed up in the first place by starting with a fresh build

joolswills, 2015-06-16:
How are you setting up multiple input devices ? From the ES config ? I doubt that works yet - the gamepad hotkey likely overrides the keyboard one or so, or we need to tweak the retroarch config. The es configuration tool is not really set up for configuring multiple devices yet (Nor has it had much testing for that afaik)

HerbFargus, 2015-06-16:
Yeah I've just been using the es config. It seems to work fine in creating autoconfigs for multiple gamepads and the hotkeys only work on the first player (which is determined by where it's plugged in as noted above), but you're right I think it just ovverrides the keyboard hotkeys, otherwise it seems to work fine.

joolswills, 2015-06-16:
I don't think retroarch supports having a gamepad hotkey and a keyboard hotkey. it's one or the other I think.

HerbFargus, 2015-06-16:
You are correct- I don't think retroarch makes a distinction so when the autoconfigs are the first thing loaded for the gamepad it ignores the hotkeys laid out in retroarch.cfg which has the keyboard configs (though it still works to control player one along with the gamepad). If retroarch treated a keyboard like a joypad with autoconfigs then the hotkeys would work but for the most part I don't see it as a major issue as most people will be using one or the other.

joolswills, 2015-06-16:
if you press the hotkey on the gamepad, you can then press f1 to enter the menu. Basically when using gamepad we enable the hotkey button, which needs to be pressed before any hotkeys work - which is usually disabled when just using a keyboard

joolswills, 2015-06-16:
it's by design anyway, so not much we can do. Cheers.

HerbFargus, 2015-06-16:
Yep its that nul binding thing. anyhow I think we can consider this issue closed, Thanks :)

McPace, 2015-06-18:
Hey guys, just finding this thread. I updated from Binary two days ago and after configuring wireless xbox360 controls the Dpad didn't work at all. Is this resolved now? Should I update from Binary or Source to test the fixes discussed above?

joolswills, 2015-06-21:
@McPace please can you try with the latest beta (but update retropie-setup via ssh before configuring emulationstation) and let me know if it works for you . thanks.

McPace, 2015-06-22:
Yes sir. I can give it a go tomorrow

joolswills, 2015-06-22:
have had confirmation this is resolved now from forum user. if there are any remaining problems we can open a new ticket.

closed 2015-06-22 by joolswills

McPace, 2015-06-22:
:( Ohhh I was almost helpful.

joolswills, 2015-06-22:
All feedback is appreciated :)