RetroPie/RetroPie-Setup #865: Couldn't open joystick #0

xand49, 2015-06-13:
Hi, Currently setting up a bar top cabinet with custom arcade controls, I'm using a Kade (not the miniconsole) plugged in via a powered usb hub. 

When I go into emulation station the controls work as configured to control the menus however they do not seem mapped by retroarch to work inside the various emulators. As such I have been trying to configure retroarch to my player one Kade and it is not being displayed in /dev/input/.

When i run the retropie setup script and use option 17 i get the following message

cannot access /dev/input/js*: No such file or directory
/opt/retropie/emulators/retroarch/retroarch-joyconfig: option '--joypad' requires an argument
Coudln't open joystick #0.
grep: /opt/retropie/emulators/retroarch/configs/tempconfig.cfg: No such file or directory

any ideas how I can get this sorted?

joolswills, 2015-06-13:
Please provide more details as to version of retropie etc - also you might do better to use the forum initially if you haven't already as there are more users there to help with issues.

xand49, 2015-06-13:
Hey, it was retropie v2.6.0-rpi2 and yes I am running a rpi2.

I will move to the forum if it is not deemed an issue

joolswills, 2015-06-13:
Plenty has changed since retropie 2.6 regarding control configuration - perhaps you could backup your current image, or use another SD card and try out the 3.0 beta 3?

xand49, 2015-06-13:
How far along is the beta? i'll be passing this system along and hesitant to have it running a beta if it isn't very stable

joolswills, 2015-06-13:
beta 3 has a few issues with its new integrated controller set up that will be fixed in the next few days - you could also try beta 2. We are aiming towards removing the beta tag as soon as this and any other problems are sorted).

joolswills, 2015-06-13:
retropie is really a rolling codebase - we constantly improve it and so the images are really just a snapshot of where we are, but currently moving towards stabilising the new stuff rather.

xand49, 2015-06-13:
Will give the forums a whirl, would imagine this was up and working in 2.6

joolswills, 2015-06-13:
I just re-read your original post and I think the problem is your controller is a HID device not a joystick. You need to configure the retroarch.cfg as keys - however the beta3 should do this automagically for you.

xand49, 2015-06-13:
what line would i need to change to what in the cfg?

or am i just uncommenting and manually switching the assigned keys for the keyboard input section?

joolswills, 2015-06-13:
first drop into a terminal and press some buttons - you will probably find your controls map to keypresses. you then need to modify the file /opt/retropie/configs/all/retroarch.cfg where the keyboard input section is to map to the keys.

The 3.0 beta3 does this all for you, based on emulationstation input.

xand49, 2015-06-13:
have installed 3.0 beta3 and set everything that way. however controls still not functioning in game, have also manually edited the retroarch cfg to match and nothing

xand49, 2015-06-13:
actually, they seem to be working on some console emulators but not on the likes of mame4all or fba which seem to be linked to retroarch?

joolswills, 2015-06-13:
depends which mame4all or fba - there is a retroarch (libretro) version and non libretro. the other ones require still manual confuguration

joolswills, 2015-06-14:
any luck ?

xand49, 2015-06-14:
managed to get consoles and mame working, just mid scraping at the mo so will try the others later.

closed 2015-06-19 by joolswills