RetroPie/RetroPie-Setup #867: Retroarch PSX emulator not communicating with certain ROMs

dlazenby, 2015-06-14:
I am using retropie 3.0 Beta 3 with a PS3 controller connected via USB.  I have loaded several playstation 1 games on the Pi and have configured the emulator to work with my controller, the issue is that some games are not receiving inputs from the retroarch emulator.  Some games (about half) work 100% fine. Are perfectly playable and I consider them some of the best emulated games on the entire retropie project. The other ps1 games however, boot up fine and the video and sound work great, but no matter what I press on the controller (or keyboard) I cannot interact with the game.  The reason I say that it is retroarch not communicating with the ROM is because the controller (and keyboard) will bring up the retroarch configuration menu and exit via hot keys just fine, but no commands get past the retroarch emulator into the actual game. 
Is there some configuration that can remedy this? Is this a known issue for some games? I would really like to play Road Rash.... Please help!

joolswills, 2015-06-14:
Did you have this on beta 2 ? Are you using the new emulationstation control configuration ? Does it help if you use the "Configure input devices for RetroArch" option from supplementary menu in retropie-setup to set up your controller after this - if so then the issue is related to https://github.com/petrockblog/RetroPie-Setup/issues/864

dlazenby, 2015-06-15:
I did not test on beta 2, I went straight from 2.6 to 3.0 beta 3 and I tried it on my backed up copy of 2.6. I got the same results, so it doesn't appear to be something that occurred because of the update.  
It does not help to configure the controller any different way.  I would like to think myself at least at or maybe above average when it comes to computers, programming, etc but nothing I would consider "intuitive" (like configuring a different way) worked to solve the issue.  Really the best explanation I can give as to the problem I am having is that it is like the psx emulator doesn't communicate with the rom (other than to boot it up).  If you would like to test it out, I downloaded Road Rash (the original, not Jailbreak or 3D; those work fine) from emuparadise. You could see if you can recreate the problem using the same ROM. Maybe it is just the ROMS I am using?  I also downloaded Monster Rancher from the same site and it has the same behavior (not responding to the controller, but I can bring up the retroarch menu using hot keys).  If there is anything else I can do to help let me know. I love this project and appreciate the hard work that has gone into it!

joolswills, 2015-06-15:
Thanks for the info. I will look into this when I next get a moment - it may be related to some other ongoing issues (which I need to fix first).

joolswills, 2015-06-15:
Missed the bit regarding it happened on 2.6 - probably an emulator bug then but I will dig further.

fkratzert, 2015-06-26:
i am facing the same problem as dlazenby. in some games everything works perfectly and in others nor the controller or keyboard are recognized as input. when i start the game in both cases my joypad (xbox 360 wired controller) is recognized as joypad but then when the game starts i just cant do anything. the only thing i can do is using the previouse (in retroarch.cfg) added commands for exiting emulators (hotbtn + exit_btn) this works but the rest of the controlls do not...would be happy for any help..

joolswills, 2015-10-15:
Is this still a problem on 3.1 ? I tried with Road Rash and it worked well. If people still have a problem I will need help reproducing it.

minimonk347, 2016-01-02:
So I am facing this very issue myself. Some of my roms recognize input when they run; some don't. They all respond to the retroarch hotkeys for savestates and exiting roms. I just updated the binaries for retropie today, and am still having the same problem. What is really strange, though, is that prior to a few months ago, this problem didn't exist; all of the roms that don't respond to input used to, and I didn't update anything, as far as I can remember. I am using a wireless 360 controller w/ xboxdrv, should that matter.

joolswills, 2016-03-26:
Quite a lot has changed in retroarch etc since this, please can someone let me know if it's still an issue with a 3.6 image.

NexxusDrako, 2016-04-01:
I'm having the same issue. Destruction Derby 2 and Brunswick Circuit Pro Bowling are two of the games that won't recognize input. Though DD2 will recognise pressing start in the intro videos, it fails to recognise any input afterwards.

wetriner, 2016-04-30:
I believe this is just a configuration issue. I tested both DD2 and Bowling, and I had control. I had a hunch this issue was due to the Pad type core option analog vs. standard. Both games responded fine on standard, but wouldn't respond on analog. DD2 did indeed recognize a button press in the intro videos on analog. It seems there's quite a lot of PSX games that can't support the dual analog controller type, so you just have to switch it back to standard for them.

HerbFargus, 2016-05-10:
@wetriner is probably correct.

closed 2016-05-10 by HerbFargus