RetroPie/RetroPie-Setup #871: Libretro-mGba -> Broke

Omnija, 2015-06-15:
Since the last update the source for libretro-mGba no longer works. 

RetroArch [ERROR] :: dylib_load :: dylib_load() failed: "/opt/retropie/libretrocores/lr-mgba/mgba_libretro.so: undefined symbol: binaryName".
RetroArch [ERROR] :: load_symbols :: Failed to open dynamic library: "/opt/retropie/libretrocores/lr-mgba/mgba_libretro.so"
RetroArch [ERROR] :: rarch_main_init :: Fatal error received in: "load_dynamic()"

joolswills, 2015-06-15:
I will compile and test and feedback upstream as needed. thanks.

joolswills, 2015-06-27:
I have forked and branched mgba before the pull request where it broke. I have opened a ticket upstream here - https://github.com/libretro/mgba/issues/5

joolswills, 2015-06-27:
This is temporary of course, until the problem is resolved upstream - so this ticket will remain open. thanks.

joolswills, 2015-06-27:
I have submitted a fix for the actual issue upstream. When it is included I will switch back to the libretro repository, and you can compile the latest version to test. thanks.

joolswills, 2015-06-28:
this is now fixed upstream.

closed 2015-06-28 by joolswills