RetroPie/RetroPie-Setup #874: Wicd-Curses not finding wireless networks

HerbFargus, 2015-06-18:
I installed retropie 3 beta 4. This may just be specific to my build or I may have a faulty wireless dongle, I tested two and they both function... wicd-curses is just having some trouble. As some people have mentioned performance issues with Wicd-Curses (https://github.com/RetroPie/RetroPie-Setup/issues/825) I'll look into working on something leaner that just changes wpa_supplicant or /etc/network/interfaces/

joolswills, 2015-06-19:
works fine here, but as you mention, maybe we should have a basic script to set up values in wpa_supplicant.conf or so.

HerbFargus, 2015-06-30:
I was looking at this script https://github.com/Death259/PiAssist/blob/master/PiAssist.sh and I tested the wifi option in his script and it worked for me. If I were to tweak it a bit as it's own script that could be called from the retropie menu would that suffice to address the cpu speed scaling issue? (also would you prefer I use dialog instead of whiptail?)

joolswills, 2015-07-01:
we use dialog yeh - the licence of that project is compatible with ours perhaps, but might be worth speaking to them - but it might just be easier to roll something simple ourselves? (for example that just allows you to enter an essid and a password and then edits wpa_supplicant.conf or /etc/network/interfaces ?

joolswills, 2015-07-02:
I have just made a small change to the script, so wicd-curses is only installed the first time the configuration is chosen. Solves the issue of it running for those just using a wired network.

HerbFargus, 2015-07-03:
yeah I've contacted him but he's yet to respond. It would be the simplest to enter the essid and password and have it edit wpa_supplicant. There may be some variations with hidden ssids, WEP, and open networks but the majority should just have wpa or wpa2. When I've got time I'll see if I can hash out a simple dialog script.

EPe9686518, 2015-07-08:
I too have had the same issues with wicd finding networks, and I am using the reliable Edimax 7811Un adapter. And I have 4 of them that I tested and a no go, but was able to configure by editing /etc/network/interfaces and it worked just fine, despite wicd not finding any networks. Please Herb, if you could do something like that script that would be perfect!!! Obviously it's no problem for people like us to get wifi working, but I am giving these retropie systems to friends who, needless to say lack computer skills in most cases, and with Kodi working so great on these now, the need for a easy to configure wifi menu would be a huge plus. One of the best things about the PIs is the portablity. It would be great to be able to switch wifi networks on the fly when people take these to other places outside of their home with a easy to use menu vs having to go into terminal and edit the /etc/network/interfaces by hand.

HerbFargus, 2015-07-18:
OK I've created a script here:
 
https://github.com/HerbFargus/Raspberry-Pi-Wifi-Scripts/blob/master/raspberrypiwifi.sh 

by modifying the PiAssist script from here:

https://github.com/Death259/PiAssist/blob/master/PiAssist.sh 

The default only had network configurations for WPA, I've added 2 more options- one for WEP, and one for Open Networks (I tested them on my Pi and they all work). The two I haven't yet addressed are static IP's and Hidden SSIDs. As of right now it uses whiptail since I couldn't figure out how to sort it out with dialog. This is just a really rough draft start and probably more redundant in the code than it needs to be (but since I lack coding experience its all I could piece together for the time being) 

Improvements needed for the future:

- Edit wpa_supplicant.conf instead of just /etc/network/interfaces as to enable multiple networks

- Add option for Static IP

- Add option for Hidden SSIDs

- Use dialog instead of whiptail so it can be added to the retropie setup script.

- Use radiolist for ssids so you can select the SSID rather than needing to type it into an input box (assuming that it shows up in the scanned ssid list)

@joolswills  What do you think?

Also as it stands right now this script can be used on it's own. To run it:

navigate to the directory you saved the script in

```
sudo chmod +x raspberrypiwifi.sh
```
needs to be run as root
```
sudo ./raspberrypiwifi.sh
```
then just follow the directions

joolswills, 2015-07-18:
I like the idea but it still might be better to do this ourselves using our existing framework and styleguide and incorporate it as a retropie-setup module. I'd be interested in getting @petrockblog thoughts though.

HerbFargus, 2015-07-21:
Just for reference, I did a bunch of testing with wifi configs as I've been working with Death259 on his wifi menu with PiAssist, and I came up with a comprehensive list of working configurations for both /etc/network/interfaces and /etc/wpa_supplicant/wpa_supplicant.conf so that if/when this is integrated to the retropie project it will be easier to configure:

https://github.com/Death259/PiAssist/issues/17#issuecomment-123132092

HerbFargus, 2015-08-02:
we can consder this resolved.

closed 2015-08-02 by HerbFargus