RetroPie/RetroPie-Setup #875: Mario Golf Advance Tour and Mario Tennis Power Tour

craigerskine, 2015-06-19:
I searched everywhere on the net for info about this issue but can't find anything.

When I try to load Mario Golf of Mario Tennis on GBA with the default settings from a fresh install of RetroPie, the games reset to the GBA loading screen when choosing a character at the very beginning.

Has anyone else seen this with any other GBA games and if so is there a workaround?

joolswills, 2015-06-19:
which emulator ? This would probably be best posted on the forum rather than this issue tracker - also please include what version of retropie etc you are running.

craigerskine, 2015-06-19:
Sorry for the post in the wrong area.

I'm using the default emulator for GBA, whatever is used if you don't change anything and just try to launch a GBA game... I'm not sure what it is offhand. I'm using RP 3.0 Beta 3, this also happened in RP 3.0 Beta 2.

Note: Both games using the exact same roms launch and play fine in other emulators as well as OpenEmu.

joolswills, 2015-06-19:
sometimes not everything works with every emulator and sometimes emulators have bugs. retropie has a selection of gba emulators which can be switched between - read the wiki for more information.

HerbFargus, 2015-06-19:
mario golf advance tour seems to work better on gpsp rather than the libretro-gpsp emulator (press x or m as a rom loads to change the emulator). Mario Golf tennis doesn't work on either default emulator, but it does work on lr-vba-next (which you would install from the experimental menu- its a bit laggy though) neither work on lr-mgba at the moment but that may be an upstream issue. I think we can consider this issue closed.

closed 2015-06-19 by joolswills

craigerskine, 2015-06-20:
Thank you for the info.

darth2499, 2019-07-23:
I know this has been closed for over 4 years now, but I wanted to point out that I have the same issue with Mario Golf where right before the character selection the GBA loading screen pops out and I can never progress any further. Running on Raspberry Pi Zero W.

craigerskine, 2019-07-23:
@darth2499 Use the no-intro romset from archive.org, that fixed all my issues.

joolswills, 2019-07-23:
Please use the forum.