RetroPie/RetroPie-Setup #876: Menu broken after updating v3 Beta 2 to Beta 3

seriema, 2015-06-20:
Technically I guess it's an EmulationStation issue, but it happened after I ran `retropie_setup.sh` then used `Update RetroPie-Setup script` and `Binary-based installation` on my Raspberry Pi 2 running RetroPie v3 Beta 2 to get the Beta 3. I didn't find any documentation for how to update beta releases so maybe I did this wrong?

This is shown when I boot: `Linux retropie 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l`

Before running the update I had nice blurry backgrounds but now it's like this on all emulator screens:
[image]

joolswills, 2015-06-20:
Looks like some settings got lost - no idea how it happened - not seen that before - were you running a custom theme etc ?. I would just start with a clean beta4 image though.

joolswills, 2015-06-20:
Updating does sometimes require manual merging of new settings etc too - but mostly retroarch.cfg and per platform configs etc.

seriema, 2015-06-20:
I had gotten a few notices after updating, but none seemed relevant. There was a "Could not install packages: wicd-curses" which seemed to be related to wifi? And BIOS files for emulators but I'm not using any of those that it prompted.

Wow there's a Beta 4 already? =D

I guessed a clean image would be the best, so it would be nice with a note somewhere for what I should think about. Besides backing up ROMs ofc. Controller config files? Scraped box arts? Wifi settings?

joolswills, 2015-06-20:
No settings should be overwritten but always good to backup - not sure why you got some failed packages, but it could be you had a networking issue when updating so it didnt install new stuff correctly - I would need logs to help more (from logs dir in RetroPie-Setup) . . There is more info on the forum and wiki.

seriema, 2015-06-20:
I ran another update (script+binaries) and now everything seems to be working! Must've been what you said, some missing package from a wifi issue. Thanks so much for the super quick response!

closed 2015-06-20 by seriema