RetroPie/RetroPie-Setup #877: Core Input Remapping Options not working in RGUI

HerbFargus, 2015-06-20:
referenced from this forum post: http://blog.petrockblock.com/forums/topic/core-input-remapping-options/#post-100250 

In theory this setting should allow one to change the default core configurations for the retropad rather than reconfiguring each controller with retroarch.cfg or autoconfigs. For example the default retropad controls for the megadrive 3 button are as follows:

[image]

but with the core remapping option you could change the retropad so that A button on the megadrive pad would be mapped in the core to X instead of Y (or any other button) without necessarily changing your retroarch.cfg or autoconfigs (so it would changed from ABC=YBA to ABC=XBA). As it stands right now, if you wanted to have Y be X, you'd have to map your X button as Y which can cause confusion for other emulators when you try to use them. -hopefully my explanation wasn't too confusing.

joolswills, 2015-07-16:
Please can you try with a new retroarch built from source ? There has been a lot of work on retroarch recently (they have been rewriting large parts), and I think this is fixed now. I just tried it and it seemed ok here.

joolswills, 2015-07-16:
Just to clarify - I've only tested this on demand in the RGUI - not saving/loading etc.

HerbFargus, 2015-07-17:
Yeah I've been meaning to test it with the recent changes. I just built a fresh image and built retroarch from binary and it works brilliantly- but my hotkeys aren't requiring me to press select first before they work- so if I just press X it enters the menu, if I press start it exits, etc. (this all happened before I made any core input remapping changes) I wonder if they changed the nulbindings with their recent updates.

joolswills, 2015-07-17:
Can confirm and a PITA as I just uploaded a new retroarch binary assuming nothing would have broken from the one I was testing which was just a few days older or so. Will look into it.

joolswills, 2015-07-17:
https://github.com/libretro/RetroArch/issues/1982

joolswills, 2015-07-17:
Fixed upstream now (At least from my initial testing). Please can you build retroarch from source again and test ?

joolswills, 2015-07-17:
I have updated the retroarch binaries also.

HerbFargus, 2015-07-17:
I updated from binary and can confirm it is now working.

joolswills, 2015-07-17:
hurray.

so back to this ticket - is the core remapping stuff working for you ok also ?

HerbFargus, 2015-07-17:
Core remapping works as well :)

closed 2015-07-17 by HerbFargus