RetroPie/RetroPie-Setup #878: Slight issue with new Emulation Station mappings

Swmbuk, 2015-06-20:
Hi,

I have a generic MY-POWER_CO.,LTD._Mayflash_Arcade_Stick that I initially set up with EmulationStation on a fresh build of 3.0 BETA 4. The feature worked great except that the direction hats were saved to the RetroArch joypad cfg folder as 'input_right = "h0right"' as opposed to 'input_right_btn = "h0right"', for each of the directional controls. The resulted in me not being able to use the joystick in RetroArch.

I went in and manually changed this and it's working great now. I didn't get this error with my iBuffalo SNES pads however, so I think it may have something to do with my stick having a directional hat.

joolswills, 2015-06-21:
Thanks for testing and the feedback. Some bugs have got through with our config generation due to lack of controllers for testing. I believe this is now fixed though in the commit above. Please drop to a terminal and try

```
cd ~/RetroPie-Setup
git pull
sudo ./retropie_packages.sh emulationstation configure
```

then restart and es should want to reconfigure the controller
if you get any errors from git pull, add sudo in front.

joolswills, 2015-06-22:
have had confirmation this is resolved now

closed 2015-06-22 by joolswills