RetroPie/RetroPie-Setup #879: Better Kodi Support

dr0id123, 2015-06-23:
Like many I'm looking to use KODI and Retropie together in harmony.  My suggestions are as follows:

1. Controller Support in KODI
2. HDMI CEC Support in KODI (adjust the KODI install script to install cec-utils package)
3. Provide an out-of-the-box option to boot directly into KODI
4. "Power Saving" options are missing in KODI -- which I suspect is causing the "Power Off System" function not to work.  This is rather annoying that one cannot shutdown the system in Kodi though the menu option is there.
5. EmulationStation HDMI-CEC support

#2, 3 -> Rather easy?
#5 -> Appears someone is working on CEC support already, might be worth to try a build? - https://github.com/Aloshi/EmulationStation/issues/421 

Retropie is awesome, please consider these suggestions.

joolswills, 2015-06-23:
Pull requests welcome :)

HerbFargus, 2015-07-10:
These are just a few ideas on how some points on your request might be implemented- I currently lack the scripting knowledge to do it myself but hopefully these ideas can at least point someone who is more programmatically inclined, in the right direction. First and foremost I think Kodi should be updated to kodi 15 (which can be installed using this guide https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=109088) and then these ideas should be implemented for Kodi 15 (which should hopefully retroactively work for Kodi 14.2)

1. There are two ways I can think of that joypad support could be implemented
 - Joy2key could be used to essentially turn the joypad keys into keyboard strokes
 - a script could be added to the inputstation controller module to create joypad configs in the keymap configs for kodi (see this page http://kodi.wiki/view/Keymap)
2.  this is a post on compiling libcec 3.0 (which is compatible with kodi 15) https://drgeoffathome.wordpress.com/2015/06/11/cec-3-0-0-on-the-raspberry-pi/ )
3. This could be added as a third option in this script: https://github.com/RetroPie/RetroPie-Setup/blob/ce8ede4defe4d2338255a202eb8e7b69be5bd7b4/scriptmodules/supplementary/autostartemustat.sh
4.  this post may have something to do with that: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=108209 but it also might be solved with Kodi 15 (I haven't tested that yet)
5. Cec support for emulationstation should ideally be implemented upstream but Aloshi got a job and emulationstation hasn't seen much development since then so it may need to be added manually to a retropie's build of emulationstation until emulationstation resumes development (again preferably with libcec 3.0 rather than the older version listed on the pull request you referenced.)

I'm not sure how many things have already been solved with Kodi 15 but at the very least it doesn't have the black screen bug when you exit kodi.

Haris1977, 2015-08-20:
+1 for all of them. I 've managed to:

1) Boot directly into kodi
3) Enable hdmi-cec
4) Pressing exit, power off, reboot to go back to ES

What i haven't be able to do is to shut down completely the system from within the kodi menu (this can be an enhancement i beleive). I have opened an "issue" here and in RetroPie forum in order to contribute any thoughts:)

https://github.com/RetroPie/RetroPie-Setup/issues/974

necrophcodr, 2015-10-05:
Regardless of how the joystick issue is solved, I believe you need to go the keymaps way. It needn't be difficult however, as a base joystick xml can be inserted, and then mapping the correct key ids to the correct keys.

However, it is important to understand that the `--enable-joystick` configure flag is required, and regardless of how automated this has to be, this flag needs to be set.

joolswills, 2016-04-09:
Repo with Kodi 16 added now for RPI - which has patches for joypad support. Requires a manual configuration currently though.

closed 2016-04-09 by joolswills

Hellcube, 2016-04-18:
Any thoughts to ad.3 i would preffer booting to kodi and switch to emulationstation from there.

Haris1977, 2016-04-18:
+1 for this

necrophcodr, 2016-04-20:
The thing about booting to kodi is that, as far as I can tell, the controller setup happens in emulator station. Booting a new device directly to kodi would yield no control.

Hellcube, 2016-04-21:
https://retropie.org.uk/forum/topic/324/start-kodi-with-emulationstation-link
This should help for start

dillbyrne, 2017-01-15:
I wrote a small program to add cec to retropie  for myself, it works with the current retropie and does not run when retroarch or kodi is running. It can run as a non root user and can be set to run at boot. Be sure to read the README first.
https://github.com/dillbyrne/es-cec-input :wink: