RetroPie/RetroPie-Setup #881: retroarch keyboard issues with mame

scottnath, 2015-06-24:
I am trying to launch games as per the instructions here:

https://github.com/RetroPie/RetroPie-Setup/wiki/How-would-I-start-from-command-line,-say,-the-SNES-emulator-by-itself%3F

but games launched this way do not recognize keyboard commands at all. I can only get F1 to work to pull up the retroarch config.

I am able to run these games just fine from emulationstation. 

Also, there is a gaminfo/nag screen when running from retroarch, but not from emulation station.

Are these bugs due to RetroPie or RetroArch?

thanks,
Scott

joolswills, 2015-06-24:
likely because you didn't pass the --config parameter with the retroarch.cfg file

You would be better to post support questions on the forum as this issue tracker is for bugs etc.

add
```
--config /opt/retropie/configs/all/retroarch.cfg
```

or point it to a specific system retroarch.cfg

closed 2015-06-24 by joolswills

scottnath, 2015-06-25:
For the record. Tried that, (`--config`) it didn't help. Looked on petrockblock and found a post from June 16, 2015 that said to reinstall retroarch from binaries...didn't help.

Seems that launching games via Retroarch from the command line brings up games which don't recognize a keyboard or a retroarch.cfg.

Might I suggest taking down this page:
https://github.com/RetroPie/RetroPie-Setup/wiki/How-would-I-start-from-command-line,-say,-the-SNES-emulator-by-itself%3F

That does not work on a fresh install of RetroPie beta version 4

thanks,
Scott

joolswills, 2015-06-25:
it does work or did when I last tried it - and the info on the page is good, so not sure what your issue is - you have a keyboard configured in the cfg ?