RetroPie/RetroPie-Setup #882: adding configs folder to ./emulationstation folder

InsecureSpike, 2015-06-24:
could this be a possible enhancement, the moving of the configs folder and contents to the home/pi/.emulationstation folder, so as this location can store/hold any/all edited retroarch.cfg files,
so as these don't get updated/overwritten/modified by updates.
and these files would override any in the original location, just like the es_systems.cfg

joolswills, 2015-06-24:
the files don't get overwritten with updates - Moving them to another location wouldn't really help and would create a lot of confusion - the .emulationstation folder is for emulationstation, and not other configs.

InsecureSpike, 2015-06-24:
ah i see, thank you for clearing up my confusion

closed 2015-06-24 by InsecureSpike

joolswills, 2015-06-24:
just fyi, if you update retropie-setup and do a new binaries install, it should not overwrite existing files but create a new config in most cases. new retroarch.cfg will be called retroarch.cfg.rp-dist for example