RetroPie/RetroPie-Setup #887: advMame rom version

amm297, 2015-06-29:
I'm going crazy trying to find a complet romset form advmame v 1.2, which version of mame roms run? I'll try 0.106, 0.139, 0.35, 0.37 and more that I don't remember

HerbFargus, 2015-06-29:
Your best bet is to use clrmamepro to rebuild the roms from a newer romset:  https://github.com/RetroPie/RetroPie-Setup/wiki/Managing-ROMs 

the .dat files needed to rebuild are clearly outlined at the bottom of the page.

joolswills, 2015-06-29:
This is not a support forum, but an bug/issue tracker - please use the retropie forum for support questions - see http://blog.petrockblock.com/forums/forum/retropie-project-forum/

Details regarding romset versions are on the wiki https://github.com/RetroPie/RetroPie-Setup/wiki/MAME

closed 2015-06-29 by joolswills