RetroPie/RetroPie-Setup #888: ARM packages for emulation station and SDL2

robertely, 2015-06-29:
Hey guys, i'm working on a public repository for use with the Pi. I'm going to feature primarily gaming, and emulation.

I made a set of packages for EmulationStation and SDL2, documenting the build scripts here:
https://github.com/Aloshi/EmulationStation/issues/462
The EmulationStation devs never replied to that issue, so these are unofficial. That said I would hope these can simplify the install process.

I'm going to push the repo public in a few days, and i'm looking for mirrors before I start telling people about it.

joolswills, 2015-06-29:
Are you asking for a mirror, or suggesting we switch to these packages ?

We currently build our own - using the SDL debian package configs as a basis with a few minor adjustments - that way we handle dependencies etc, and any customisations we might need (as we have with SDL1 for example).

Script here

https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/sdl2.sh

robertely, 2015-06-29:
I suppose both, I'd like to host the packages in a real repository so you don't need to download blobs, or install it out side of apt. 

If that's to be depended on, it would need mirrors, and more than one official maintainer.

joolswills, 2015-06-29:
I don't know that we would want to use 3rd party SDL packages as we wouldn't be able to patch/customise and push out updates ourselves. Our emulationstation is also a custom fork, so we don't use the standard build.

We don't have a real repository, but only because we only have a couple of real debian packages, and we can just download/install them via the update script so it's not really worth the additional maintenance/work. Also if/when retropie works on other hardware we will need to do this differently again, which is easier with the simplicity of the current setup rather than having to have another repository for that case etc.

joolswills, 2015-06-29:
Your repository would definitely be useful for raspbian users though who just want to install a stock sdl/emulationstation.

robertely, 2015-06-29:
Didn't realize your emulation station was that customized. Are the binary builds different or can your changes be merged in via a supplemental package?

joolswills, 2015-06-29:
the binary is custom - https://github.com/RetroPie/EmulationStation - includes additional controller configuration stuff.

joolswills, 2015-07-01:
Perhaps you could try and get an SDL2 built into the rpi repositories ?

joolswills, 2015-07-01:
I'll close this issue for now though, as it's not an issue as such. cheers.

closed 2015-07-01 by joolswills