RetroPie/RetroPie-Setup #891: Xbox Controller Problem

LucasRoot, 2015-06-30:
Hi, when configuring the keys of my Xbox 360 controller, was giving some bugs.
1 - When try to set up the right trigger the RetroPie just skipped
2 - I can't move in the games, like, I can jump, shoot, but move no.

Thanks and sorry for my bad English

HerbFargus, 2015-06-30:
are you using retropie 3 beta 4? sounds like this issue https://github.com/RetroPie/RetroPie-Setup/issues/864. update your setup script, and reinstall emulationstation and see if that fixes your problem.

joolswills, 2015-06-30:
I've had problems with shoulder buttons with the configuration too - you can skip them for emulationstation and try to use the retroarch input configuration option from the supplementary menu.

joolswills, 2015-07-02:
try installing the xbox 360 userspace driver from the supplementary menu of retropie-setup and rebooting. This fixes controller issues with triggers on my original xbox controller.

LucasRoot, 2015-07-02:
Hi, I tried your fist solution, didn`t work, I will try your send solution.
Thanks joolswills!

LucasRoot, 2015-07-04:
Hi, it won't work :(, I can't find the retropie 3.0 BETA 2 to download, I will continue search.
Thanks for all the help

Kolin, 2015-07-21:
I'm still seeing this problem in Version 3.0 RC 1

joolswills, 2015-07-21:
did you install the xbox 360 userspace driver ?

joolswills, 2015-07-21:
Also you didn't specify which problem - please be more verbose with issue reporting.

LucasRoot, 2015-07-23:
Yes, I did.
With the new update of the beta 2 of the 3 version, the controller jump some steps when setting up in the main screen when booting. And after the config, in the game I can't move

joolswills, 2015-07-24:
I was answering @Kolin :)

@LucasRoot please test with a fresh image of 3.0 rc1 (and install the userspace driver first). thanks.

Kolin, 2015-07-24:
@joolswills I was referring to the problem mentioned in the first comment.
Yes, I've tries the userspace driver.
The problem still exists while trying to set up a controller via emulation stations interface.
The retroarch controller set up script works fine.

Cheers,
Colin

joolswills, 2015-07-24:
@kolin The left right triggers should work in emulationstation after setting up the userspace driver (and rebooting). What is displayed in ES at the bottom of the controller configuration window when setting up the controller after this is done (the name of the device).

joolswills, 2015-07-24:
(Basically when you press a button to configure the controller it displays the name - what does it display)

Kolin, 2015-07-24:
Thanks, 
I'll try this again with a fresh install when I get home from work.

njradford, 2015-08-26:
I'm encountering similar issues with the shoulder buttons. The initial joypad config works great until the right trigger is mapped, at which point it will fail to recognize the input and prompt me with a "Hold to skip" countdown. I've tried looking around to a vanilla/functioning xbox360.cfg file, but to no avail.

joolswills, 2015-08-26:
You need to install the xbox userspace driver first from the supplementary menu.

njradford, 2015-08-26:
Done and working. Thank you.

sellabus, 2015-09-24:
I'm having this exact same problem, using a Logitech F310 controller.
This problem exists before and after installing the Xbox userspace driver from the supplementary menu in RetroPie-Setup.
The name of the controller is displayed in ES as "Logitech Gamepad F310" both before and after installing the Xbox userspace driver.
The controller is set to "X mode" (meaning that it is operating as an xbox 360-compatible xinput device)
This is a fresh install of RetroPie.
I am using a Raspberry Pi 2 with RetroPie 3.0 installed. I have reinstalled this image 5 times and redownloaded it twice, and the problem persists.
I have a USB keyboard and WiFi dongle plugged into the Raspberry Pi along with the controller.

joolswills, 2015-09-24:
I don't think you should be using the xbox userspace driver on that controller - but see this thread in regards to your problem - http://blog.petrockblock.com/forums/topic/logitech-f310-controller-cant-assign-control-to-top-right-trigger/#post-104784

closed 2015-09-24 by joolswills

NormlDotOrg, 2016-01-16:
If anyone is still having this issue (I am as of RetroPie 3.3) just map the controls (wired xbox 360) like normal and when you get to "ok", use the dpad navigate back up to "lower right". Select with "A", it will say "press anything" and map to lower right trigger. Then go back down to "ok" with the dpad once more and strike "A". Worked for me.

betam4x, 2016-08-04:
This issue still exists in 4.0rc1.  The workaround @NormlDotOrg listed does work, but it's still a pain.

joolswills, 2016-08-04:
it is unlikely there will be a fix anytime soon - also see https://github.com/retropie/retropie-setup/wiki/Logitech-controllers#l2r2-top-triggers which is related.

joolswills, 2016-08-04:
I don't have this problem with the userspace driver, just the kernel xpad driver. However, the userspace driver has problems on the most recent kernels.

Debiruman, 2016-11-15:
Hello, I just installed Retropie those last days with the latest release (november 2016) for the Pi 1B, and I had the same right trigger issue with the XBox controller (wired).

I figured 2 ways to fix this, the first one (thanks for NormlDotOrg a couple of posts earlier for that one) is when you map the controller, instead of validating by pressing "OK" at the end of the process, go back to the right trigger maping line, and press "A", then you're asked to press a button and you finally can press the right trigger which will be properly recognized, then, go all the way down and press "OK" to complete the process.

The other way is by editing the MicrosoftX-Box360pad.cfg file, which is found at that location : 

/retropie/configs/all/retroarch-joypads/MicrosoftX-Box360pad.cfg

The most comfortable way for me was to remotly access the file from another computer in the network, you just have to click your way to the file, while it can be quite cumbersome to do it directly from the Pi, unless you're at home with Linux environment.

So, you open that file, and search for that line :

input_r2_axis =

The right trigger identifier is +5, so, after the "=" sign, write "+5", so the line looks like this :

input_r2_axis = "+5"

(note that there's a space between the = and the "+5")

By the way, while we're at it, I strongly recommand you to change the hotkey button, which by default is mapped to the button "6" (which is either the start or back button), it can be troublesome with some emulators, especially the Game Boy emulators, because in a game like "Zelda - Link's Awakening", in order to save your progress, you need to press simultanously A + B + select + start, which with the default hotkey mapping will result in exiting the emulator without saving anything.

So, in the same configuration file, find that line :

input_enable_hotkey_btn = "6"

And replace the 6 by any non-used button, I suggest the middle button of the gamepad, the one with the X surrounded by the leds, its number is 8, so the line should look like this :

input_enable_hotkey_btn = "8"

Save the file, restart Emulation Station and voilà, it should be all good :-)

yanx007, 2016-11-19:
Thank you very much Debiruman. I have Xbox One controller also. Which code
should be for input_r2_axis? and on what configuration file?
Tell me whenever you can.

But thank you very much to take the time to write it all.

Cheers!

Yan.