RetroPie/RetroPie-Setup #892: msx not run

amm297, 2015-07-01:
I intall msx emulator through retropie_setup.sh, but when launch a game it don't run and come back to emulationstation.

I test games .rom, .msx and .msx2 and none of that files run, I need to install other emulator?

joolswills, 2015-07-01:
which fmax emulator ? what are the contents of /tmp/runcommand.log ? what version of retropie ? did you use an image, or install on top of raspbian ?

amm297, 2015-07-02:
I install de 2.6 retropie version intalled by image downloaded from "http://downloads.petrockblock.com/images/retropie-v2.6.0-rpi2.img.gz", and after that update the retropie_setup.sh, install the lr-fmsx and it doesn't work, so i try to install the openmsx, it works but i can't play with a controller ,  the /tmp/runcommand.log is empty...

amm297, 2015-07-02:
I reintall hole system and it seems to work now, thanks

closed 2015-07-15 by joolswills