RetroPie/RetroPie-Setup #896: Weird/corrupted fonts in EmulationStation

C-Fu, 2015-07-03:
Running on the latest Raspbian + RetroPie beta 4

http://i.imgur.com/rsBZchF.jpg

Should also note that I did this before installing RetroPie to remove all traces of x11 + desktop:

$ sudo apt-get install deborphan
$ sudo apt-get autoremove --purge libx11-.* lxde-.* raspberrypi-artwork xkb-data omxplayer penguinspuzzle sgml-base xml-core alsa-.* cifs-.* samba-.* fonts-.* desktop-* gnome-.*
$ sudo apt-get autoremove --purge $(deborphan)
$ sudo apt-get autoremove --purge
$ sudo apt-get autoclean

joolswills, 2015-07-03:
Unable to reproduce, but you are removing lots of stuff that retropie-setup will need and re-install - if you want a cut down raspbian why not start with the retropie image which has unneeded stuff already removed ?

InsecureSpike, 2015-07-04:
i've notice this can happen, when switching from themes, if themes contain different font sets, so not so much as a corruption as maybe an unavoidable glitch, maybe

amigagamer, 2015-07-04:
Ive had similar problems with display, especially fonts.  For me it was resolved by increasing my GPU memory.  Try setting 128 or 256mb to graphics and see if it resolves the issue.

joolswills, 2015-07-05:
Good point.

C-Fu, 2015-07-06:
Sorry for the late reply. 

"why not start with the retropie image which has unneeded stuff already removed "

[@joolswills]
I couldn't because I'm using berryboot, and berryboot's RetroPie 2.3 image doesn't seem to be working if I update it to current (that dreaded missing system error).

So I had to use the Raspbian image, remove desktop + x11, and install retropie manually.

But once I tried setting gpu_mem to 256 from 128, problem solved! :dancers: Cheers [@amigagamer]!

joolswills, 2015-07-14:
closing as resolved.

closed 2015-07-14 by joolswills

davorb, 2015-09-06:
This just happened to me. These are the things I did to get the issue:

1. ```dd``` the sd-card image to a card and boot up retropie
2. Put in a usb disk and let retropie crate the different folders
3. In the folder that lets you move settings to the pi, extract [the themes](http://blog.petrockblock.com/retropie/retropie-downloads/theme-archive-for-emulationstation/) to a folder called ```emulationstation/themes```. 
4. After I restarted I started experiencing the issues with the fonts.

joolswills, 2015-09-06:
This issue is closed - it's likely due to the same problem as above - so try increasing gpu_mem - I would recommend posting on the forum first, and opening a new ticket if your problem cant be resolved.