RetroPie/RetroPie-Setup #899: Rom based retroarch.cfg not loading?

biscuits99, 2015-07-04:
This could be me, but when I set a rom based retroarch.cfg (pressing x as the rom starts), whilst it correctly saves it, e.g.
/home/pi/RetroPie/roms/megadrive/Sonic The Hedgehog (W) (REV01) [!].gen.cfg

It doesnt seem to load it, as the settings arent taking hold.

the /tmp/runcommand.log says
RetroArch [ERROR] :: config_load_file :: Failed to append config "/tmp/retroarch.cfg,/home/pi/RetroPie/roms/megadrive/Sonic The Hedgehog (W) (REV01) [!].gen.cfg"

dankcushions, 2015-07-10:
Same issue for me!

joolswills, 2015-07-11:
It's a quoting issue I think - due to rom name - I will sort it over the weekend. thanks for the heads up

dankcushions, 2015-07-11:
FYI my error happened with a file called 'donpachi.zip.cfg' - same error message as floob but no spaces/special characters in the name. Might be a separate issue after all!

joolswills, 2015-07-11:
are you running the latest retroarch - there was an earlier bug that happened when appending configs which should be fixed now.

biscuits99, 2015-07-11:
Could I confirm if I test some more, if RetroPie 3 beta 4 is ok to test with, or should I update retroarch binaries from that version?

joolswills, 2015-07-11:
ok. It seems that retroarch has changed the appendconfig delimiter - which is somewhat annoying as I need to update runcommand, but if someone updates runcommand but not the retroarch binary that will break too.

joolswills, 2015-07-11:
I opened up an issue regarding this as I don't think the current solution is great - https://github.com/libretro/RetroArch/issues/1945

joolswills, 2015-07-11:
fixed for now - you will need to update retropie-setup from the gui (or from commandline then re-run runcommand install). Hopefully a better system is implemented upstream (I may have a look at possibly creating a patch)

closed 2015-07-13 by joolswills

biscuits99, 2015-07-14:
Thanks! I'll test it out in the new beta. Am I right in thinking the appendconfig for the per rom retroarch.cfg will make this the last file read/applied (after the system specific and /all/retroarch.cfg)?

joolswills, 2015-07-15:
yes.

same31, 2015-07-15:
Hi joolswills,

I fetched the new retropie-setup but I do not find how to perform the runcommand install you mention.

Could you help me please ?

Thanks

same31, 2015-07-15:
I managed to get it installed by reconfigure it into the 'Setup / configuration' retropie-setup menu entry, and selecting the default option.

Thanks !

HerbFargus, 2016-01-03:
Hate to revisit this (might just be my incompetence) but the custom configs don't seem to be loading on a per rom basis (this is on a fresh 3.3.1 wheezy image).

is there a specific syntax for getting the rom specific config to append? the file name is created just fine but it seems that none of the settings are carrying through. For example I am trying to switch the player order for players one and two (it works fine when I manually edit `/opt/retropie/configs/snes/retroarch.cfg`):

with snes9xnext I'm playing super mario kart and the config file created is:

`Super Mario Kart (USA).sfc.cfg`
 
I've tried both 

```
input_player1_joypad_index=1
input_player2_joypad_index=0
#include "/opt/retropie/configs/all/retroarch.cfg"
```

and 

```
#include "/opt/retropie/configs/all/retroarch.cfg"
input_player1_joypad_index=1
input_player2_joypad_index=0
```

Am I missing a syntax or something? or do I need to manually edit emulators.cfg to append per rom configurations? (or did retroarch break/change it again with their development?)